Ultimaker/Cura #6478: [4.3] Printer and Materials parameters suddenly change or disappear

benjacusu, 2019-10-04:
**Application version**
Cura 4.3

**Platform**
Windows 10

**Printer**
Custom FFF Printer whith 1 extruder

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Ellecross, 2019-10-04:
@benjacusu Did you accidentally submit this ticket unfinished? 
Can you please complete it, otherwise, we're going to close it.

Ghostkeeper, 2019-10-07:
(benjacusu has edited his ticket to complete the template.)

Basically what the report seems to be, is:

* Add a custom material.
* Change any setting.
* The material disappears?

I don't see that happening though. I duplicated PLA and renamed it to BLA. Then I changed the infill density to 42%. BLA was still selected.

deathfrozen, 2019-10-25:
Also found this problem. Can't maintain the correct diameter of plastic in the extruder settings - resets to default.

closed 2020-06-19 by no-response[bot]