Ultimaker/Cura #6482: Updated to 4.3 and no third option for align to build plate.

MrJabu, 2019-10-04:
I am not seeing the third option add to Cura 4.3

[image]

smartavionics, 2019-10-04:
The availability of this feature depends on the level of OpenGL support your computer provides. If the preview screen uses the old "compatibility mode" with a single slider on the right to select the layer of interest rather than 2 sliders (one on the right and one on the bottom) which you get with the full layer view, then it indicates the OpenGL support is not sufficient to give you the align to build plate feature.

closed 2019-10-07 by nallath

rburema, 2019-10-08:
In the case you _do_ have a graphics card that's capable of OpenGL 4.1 (or higher):
 - (often the case on laptops) check if you're not accidentally rendering with the Intel chip on the main board
 - (might help) update your drivers