Ultimaker/Cura #6483: Cura 4.3.0 Ender 3 Print bed is off

Immatture, 2019-10-04:
[image]
The Y-axis is off by several millimeters and makes it so I am unable to use my full print bed.

Immatture, 2019-10-04:
[image]
This is the image of it in Cura

Ghostkeeper, 2019-10-07:
So the Y=0 coordinate is actually outside of the bed? I don't think this is really a problem in Cura but rather by manufacturing or assembly. Or do all Ender 3's have this problem?

Immatture, 2019-10-07:
It never did this before, just in the Cura 4.3 update

Ghostkeeper, 2019-10-08:
Here is a zip archive with two 1cm cubes sliced in the front-left corner of the build plate on 4.2 and 4.3: [cubes_ender3.zip](https://github.com/Ultimaker/Cura/files/3701502/cubes_ender3.zip)

In 4.2, the coordinates of the g-code range from 0.22 to 9.78. In 4.3, the coordinates of the g-code range from 0.2 to 9.8. This is due to the change in the default line width. But it's only a 0.02mm change. I don't think you took that picture with an optical microscope! Since the g-code didn't change between the two versions, I don't think that the update to 4.3 did anything to your printer.

alexchiolo, 2019-10-11:
Possibly a loose Y endstop that is moving back a little bit every time the printer homes to it?

oofiksoo, 2019-11-02:
default ender firmware has the nozzle rest off of the bed at absolute home. this is corrected by adjusting and re-imaging the firmware. All ender 3's will do this. this is not caused by assembly issues. The nozzle is off the bed during warming, so that any oozing is not on the bed during warm up. I have correctly set my marlin firmware offset after homing to -7, which will place the nozzle in the correct location when using cura. in essence, this setting in marlin is telling the printer that end stop is -7 before 0, so after hitting end stop, go forward 7 to hit the edge. Depending on use of glass such as a "ultrabase" which has a 1mm "bevel" on all edges, you may need to set this to -8 to avoid this bevel. if you are using the removable bed, I caution anyone attempting to use there "full bed size"
as you must account for the clips. as to the "loose endstop". that could be true as well, but I can fully and 100% confirm that the nozzle at "endstop" position is off of the front edge on ALL enders. if you have marlin correctly configured you will not notice this, as it will automatically move the bed up after homing to the endstop position.

oofiksoo, 2019-11-03:
btw - I love how the cura guys always throw the "assembly" thing out there. just for FYI, the lower construction is pre-assembled. while I cannot attest to Chinese QA methodology's and effectiveness, it should be mentioned that this is not a component of the ender that is being built each time by the end user, and always immediately blaming the construction of an ender3 seems to come off in my opinion as an attempt to discredit the ender, and subtly knock "kit" based printers compared to your all mighty expensive pre-built ultimaker printers.

Ghostkeeper, 2019-11-04:
Sorry, the comment about "assembly or manufacturing" was not intended to depreciate home-assembled printers. I was deliberately including manufacturing as well. In any manufacturing there will be some variation in the size of the components, and in any assembly there will be some play. I don't think that home-assembled printers are any worse than factory-assembled, since the type of user that assembles 3D printers at home is often very technical and cares about the accuracy of their product more too.

It's just that the zero position being off could well just be a physical thing.

Cura is basically assuming that the [0,0] position is always inside the build volume by definition. If it's not, the firmware could shift the [0,0] position such that it is. But I guess any assumption is dangerous with so many printers on the market.

If we need to change something, please test with stock firmware at least! If people have modified firmware, they can choose to have modifications that work with their slicer, but not everyone can make the choice to use modified firmware.

alexchiolo, 2019-11-04:
Hmmm @oofiksoo my ender 3 Pro definitely had the factory end stops set to home inside the print bed.

Anyway, more to the point of @Ghostkeeper 's last comment, I think the most flexible option would be to separate the machine settings so that Xmin, Xmax, Ymin, Ymax of the print bed can be specified separately, rather than the current width and depth with assumption of 0,0 being inside the print area.

This insures cura will always slice within a safe print area the user is comfortable with, as this is typically smaller than the physical print bed or full XY travel due to the presence of corner clips/crews on the bed and the physical construction of the XY travel system.

oofiksoo, 2019-11-04:
@Ghostkeeper I deleted the post, because it was not appropriate. My apologies to the thread.

closed 2020-06-19 by no-response[bot]