Ultimaker/Cura #6486: Missing layer depending on Initial Layer Height

Marcel-a, 2019-10-05:
**Application version**
4.2.1

**Platform**
Windows 10 Edu 64bit version 1903

**Printer**
Creality Ender-3
[curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/3693770/curaproject.3mf.zip)

**Reproduction steps**
Slice the model in the attached file

**Actual results**
Layer 21 has only a small part filled.
[image]
Layer Height: 0.24mm
Initial Layer Height: **0.16**mm

**Expected results**
All layers are completely filled.
[image]
Layer Height: 0.24mm
Initial Layer Height: **0.2**mm

Ghostkeeper, 2019-10-07:
Setting the line width to 0.04mm and outer wall inset to 0mm reveals that the slicing stage got the result correctly, but something is going awry with that shape at that layer when doing an inset of about 0.2mm for the outer wall. We may have a Clipper bug here.

A workaround for now is to set the Outer Wall Inset to 0.01mm.

It's worth mentioning maybe that in your project file the entire first layer also becomes empty because the model is too thin to print there. Maybe something you want to look at.

Werebear93, 2019-12-09:
I might have the same problem, I think. For what I've discovered: The missing layer is not really missing, but rather added under the first layer (negative z). Here's a screenshot:
[image]

Ghostkeeper, 2019-12-10:
This is a known problem that occurs very rarely. It's a visualisation problem that's only in the front-end. Sometimes one of the layers gets assigned Z=0 for some reason. We haven't been able to fix it because it only happens once in maybe 1000 slices. If you reslice it goes away, and it's not actually at Z=0 in the g-code.

Unless you can show me the part of the g-code where it goes to that Z coordinate? Then it might be a different issue.

Werebear93, 2019-12-10:
I didn't verify it and just assumed it was an error in the slicing engine. As you said, it disappears when you reslice. 😌 thanks for the quick answer anyways. I might look into the gcode when this happens again

rburema, 2020-02-07:
Devs: On our internal backlog as CURA-6893

rburema, 2020-02-07:
@Marcel-a When I try the checker from the Mesh Tools plugin on your object, it says the object is not watertight and consists of 3 sub-meshes. In general, we only support watertight models.

In this specific case, this can be fixed by (downloading the Mesh Tools plugin via Marketplace), selecting the object, then 'Extensions -> Mesh Tools -> Split Model Into Parts'. (This requires 'auto drop to buildplate' in the preferences to be off, otherwise the constituent parts of the model will _all_ rest on the buildplate).

If possible, you could also have your CAD software merge the model into a single mesh before loading it into Cura.

closed 2020-02-07 by rburema