Ultimaker/Cura #6490: [4.3] "Zoom toward mouse" and "Invert zoom" interact entertainingly

drifkind, 2019-10-06:
**Application Version**
4.3.0

**Platform**
MacOS Mojave 10.14.6

**Printer**
Creality Ender-3

**Steps to Reproduce**
In Preferences, under General/Viewport Behavior, set "Invert the direction of camera zoom" on and "Zoom toward mouse direction" on. Load a model. Zoom in or out, then use the right mouse button to (try to) orbit around it.

The view orbits around a point that appears to move in space opposite to the zoom direction. As you zoom toward the model, the point moves away from you, and vice versa.

Another possibly related hiccup. Set "Invert zoom" on and "Zoom toward mouse" off. Now zoom all the way out. You can no longer zoom in.

Ghostkeeper, 2019-10-07:
Not reproducing the related hiccup, but I see the original issue yes. I think after disabling the zoom towards cursor, we might as well just set the origin of the camera somewhere in front of the camera to fix this issue.

closed 2019-10-08 by Ghostkeeper

edm00se, 2019-12-08:
I'm still not able to zoom toward mouse in 4.4, which I first noticed in 4.3.

commit f8449796 (2019-10-08): Reset the camera origin when uninverting zoom direction

If zoom-to-cursor has been activated the camera origin will end up behind the camera. This resets it.

Fixes #6490.