Ultimaker/Cura #6492: Cura 4.3 Not Loading with Intel HD Graphics 4000

bmizes, 2019-10-06:
I've never used any version of Cura previously.  I've tried loading Cura 3.4 on my Acer Aspire M laptop with a Intel HD 4000 graphics adapter.  The program installs correctly, but when you attempt to start it, it loads machines and then tries to start the interface, at which time it crashes and you get an error message that "Cura has stopped working".  I've seen numerous reports of this on forums with this Intel graphics 4000 chip.  I'm on Windows 8 and I have also tried every compatability setting that's available to me with no success.  Do you have a fix to get Cura 4.3 to work with this graphics chip?  Thanks.  Cura log file is attached.
[cura.log](https://github.com/Ultimaker/Cura/files/3694949/cura.log)

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Ghostkeeper, 2019-10-07:
Did you try updating your graphics drivers?

bmizes, 2019-10-07:
Yes, I verified I had the latest graphics drivers.  This is a known
problem.  The question is whether there will be a solution

Barry Mizes
7001 Washington Ave.
University City, MO 63130

home: 314-7211841
cell: 314-4840504

LipuFei, 2019-10-08:
Hi @bmizes, sorry to tell you that we decided to defer this issue. The problem is probably in the Intel graphics driver so we can't do much about it. We also don't know if there is a solution for this.

lbause, 2020-01-10:
> Did you try updating your graphics drivers?

That today has worked for me following these steps:
1.  uninstall gfx driver 
2. reboot
3. make sure only the windows default windows display driver is loaded (Cura will not start, missing ogl2)
4. download and install latest intel hd4000 drivers from intel 
5. start cura 4.x worked for me after trying almost everything

bmizes, 2020-01-10:
I have the latest drivers installed.  I don't see much point in
uninstalling and reinstalling the same set of drivers.

Thanks.

lbause, 2020-01-10:
same for me. That was my last resort. And it worked. Windows was always claiming that i do have the latest and greatest drivers installed. I found an article on how to fix open gl isssues in Windows 10 which was recommending the above steps. Worked for me.

bmizes, 2020-01-10:
Ok.  Maybe I'll give it a try this weekend.  Thanks.

bmizes, 2020-01-11:
I gave it a shot but still same result - Cura won't fully load.  Thanks for
the advice.

lbause, 2020-01-11:
too bad. it made my day yesterday. as i was trying since two weeks to run it. None of the versions from 3 to 4 did work.

CheloURY, 2020-01-14:
I have the same problem with my HP notebook with Intel HD4000 graphics chip,  I have the latest drivers installed...I've tried other versions from 3.6 to 4.4 ..but none of them works.

Ghostkeeper, 2020-01-15:
Weird how these reports started happening around October but all older versions of Cura didn't work either apparently. It really must be something in that driver.

bmizes, 2020-01-15:
Absolutely.  Ultimaker support told me it was a driver issue but they had
decided not to try to "fix it".  Maybe someone will stumble upon a solution
at some point.

Ghostkeeper, 2020-01-15:
I think our support department guessed that it's not a bug in Cura but in that driver, and I agree with them. Updating the driver should not break older software. I'll close this ticket but of course you can keep discussing it, especially if you find a workaround.

closed 2020-01-15 by Ghostkeeper

fieldOfView, 2020-01-16:
@bmizes: are you checking for the latest driver with your computer manufacturer or windows update, or with Intel? Intel will probably have newer drivers.  

@lbause: what driver did you download that fixed it (and where from)?

CheloURY, 2020-01-16:
fieldOfView, I believe that the latest drivers don't work with HD4000 and Cura yet. Nobody could resolve the problem in this forum so Ghostkeeper closed it yesterday.

fieldOfView, 2020-01-16:
For the record, I am not affected by this issue, but I am trying to help people fix it on their system like @lbause seems to have been able to.

"latest drivers" is not a very precise description; latest according to who? HP? Microsoft (windows update)? Or Intel?

CheloURY, 2020-01-16:
In my case, HP notebook, WIN10 64bits Pro, Intel HD4000 are all updated. Too many people tried updating drivers without success too. I understand you are helping, I'm in the same way.

bmizes, 2020-01-16:
I'm using the latest driver according to Intel.  I even used the Intel
Driver Support & Support Assistant tool to scan my computer and confirm I
have the latest driver.  I duplicated what @lbause did but Cura would not
work for me.  I'm using Prusaslicer and Ideamaker and they are serving my
purposes

CheloURY, 2020-01-16:
@bmizes  I did exactly the same. I used the Intel app and scanned my pc but it didn't work. Now I'm using the Makerbot app to evaluate material etc.

lbause, 2020-01-17:
t

> For the record, I am not affected by this issue, but I am trying to help people fix it on their system like @lbause seems to have been able to.
> 
> "latest drivers" is not a very precise description; latest according to who? HP? Microsoft (windows update)? Or Intel?

True, so latest driver for me means: win64_15.33.48.5069 which i have donwloaded as zip from:
[win64_15.33.48.5069.zip](https://downloadmirror.intel.com/28799/a08/win64_15.33.48.5069.zip
)
Installed on Win10 Home(1909) (64bit)

By the way, in the meantime there was an update on intel.com to 15.33.49.5100 
My Notebook is a HP Elitebook 8470p

For me the update was always claiming that the newest driver is already installed. So i have made it by hand:
1, complete deinstall
2. reboot
3. install driver from ZIP file
4. start CURA (worked w/o reboot)

Before i did this Cura was always crashing (and referring to some exception in Intel HD4000 GFX driver dll) on startup. I have tried this as well on a Samsung NB using the same GFX adapter.

bmizes, 2020-01-17:
I'm running Windows 8 so the latest driver is from 2015.  Hopefully for
those that are using Windows 10, a driver update solves the problem.  While
Cura is definitely the #1 used slicer, it is by no means the only capable
one available.  I've had good luck with Prusaslicer and Ideamaker.
Changing slicers was a much more economical solution than buying a new
laptop.

Ghostkeeper, 2020-01-20:
> For me the update was always claiming that the newest driver is already installed.

This is because the vendor of your computer sometimes sits in between and waits with releasing Windows updates until they tested the update with their laptops.

headcra6, 2020-06-06:
I've sorted out this issue on my Windows 10. I have a similar case on my laptop HP ProBook 6570b. The latest available driver for the laptop provided by HP is 9.x and published in 2015. I found the latest driver from Intel which was populated on 2020:
https://downloadcenter.intel.com/download/29473/Intel-Graphics-Driver-for-Windows-15-33-?product=81499

The following are steps which should be done:
1. Download the latest driver
2. Start the installation. When you receive the message that this driver is not suitable for your device do not close this message.
3. Open explorer and in address line type %Temp% and press enter.
4. In the temp folder order files by date modify and find the latest installation log file named as Intel(R)_Graphics_Driver_Software_xxxxxxxxxx.log and open it
5. In log file find the place where the installation extract files, in my case it was C:\ProgramData\Package Cache\62E7955FDFA37FB5729BDF259F96B01DB56D1573\
6. Copy all content of this folder somewhere in your location eg. C:\temp
7. Open the Device Manager
8. Find the Video adapters -> Intel(R) HD Graphics 4000 right-click -> Update Driver
9. Press Search on my Computer
10. Press Choose the driver from the available...
11. Press Install from disk
12. Find your folder with backup files from the step 6. and choose C:\Temp\Graphics\igdlh64.inf
13. Confirm the installation

After this procedure, I've got the latest driver from Intel installed on my device and Cura started successfully

Hope it helps

tsimola, 2021-11-22:
> I've sorted out this issue on my Windows 10. I have a similar case on my laptop HP ProBook 6570b.....
> 
> The following are steps which should be done:

This works on my HP Ultrabook (Folio 9470m) with Intel 4000 graphics card! Thanks!