Ultimaker/Cura #6493: Artemis 300 2nd Extruder not an option

pjkoths26, 2019-10-07:
<When setting up printer settings for SEEMECNC Artemis 300 there is only a one extruder option available.  Artemis has a dual extruder

nallath, 2019-10-07:
That's because when the profile was added, the only version of the Artemis that was there was a single extrusion printer. 

We have to rely on what is contributed when it comes to third party printers (eg; Anything that isn't an Ultimaker). So, unfortunately, the answer is that pull requests that fix this issue (add a dual extrusion variant for the Artemis) are welcome.

pjkoths26, 2019-10-07:
Your answer is only applicable to this new version of Cura.  Previous versions I had the option to select one or two extruders from the drop down menu.

fieldOfView, 2019-10-07:
Then you were using a modified version of Cura, and did not carry over the modifications (properly).
The definition for the Artemis that shipped with Ultimaker Cura has always only had a single extruder:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/seemecnc_artemis.def.json

What previous versions did you use? Perhaps the proper definition for the Artemis can be copied/updated from that.

pjkoths26, 2019-10-07:
Then you just identified a new problem as all modifications DID transfer over less the extruder modification. I went back to 3.6.0

fieldOfView, 2019-10-07:
I went back to not caring about this issue.

pjkoths26, 2019-10-08:
Thank you for showing your true dedication to Cura that apparently is nil.

Since you don't care about the software why don't you resign and let people who truly care and want to work on the software do the job.

I don't see that happening as I can see you are an arrogant piece of SHIT!

Regards,

Pete

Ghostkeeper, 2019-10-08:
Cura 3.6 also didn't have any 2nd extruder though. Here is a screenshot of Cura 3.6's machine settings screen when the Artemis 300 is selected:
[image]

And you can see in our [source code for 3.6](https://github.com/Ultimaker/Cura/tree/3.6/resources/extruders) that there was only a file for `seemecnc_artemis_extruder_0.def.json`, not any `..._extruder_1` or anything.

Maybe you could add a second extruder definition, then? We'd need to know stuff like the nozzle offset and the material diameter. It needs to be a file similar to [this one](https://github.com/Ultimaker/Cura/blob/master/resources/extruders/seemecnc_artemis_extruder_0.def.json).

nallath, 2019-10-08:
> Thank you for showing your true dedication to Cura that apparently is nil. Since you don't care about the software why don't you resign and let people who truly care and want to work on the software do the job. I don't see that happening as I can see you are an arrogant piece of **** Regards, Pete

I've edited your message, since there is really no need to start calling people names. 

Also, fieldOfView is a comunity member that is spending his free time fixing things. I think the reaction is quite over the top. He (correctly) explained to you that what you claim was not there in any version shipped by Ultimaker. The commit that added it in the first place (https://github.com/Ultimaker/Cura/commit/aba3795d0b51b3ca8fa5307bd380a0a8fade6aff) explicitly states that it only had one.

pjkoths26, 2019-10-08:
>     Also, fieldOfView is a comunity member that is spending his free time fixing things. I think the reaction is quite over the top. He (correctly) explained to you that what you claim was not there in any version shipped by Ultimaker. The commit that added it in the first place (aba3795 https://github.com/Ultimaker/Cura/commit/aba3795d0b51b3ca8fa5307bd380a0a8fade6aff ) explicitly states that it only had one.

Mr Van Kessel -

An arrogant reply from that person like I received was totally uncalled for and also "Over the Top".  If my response seemed to be name calling it wasn't meant to be, I only described the a person who would respond such as he did to me or anyone.  If he is such a valued contributor then you might consider filtering responses he provides like you did with mine.  His response to my request only reflects negatively on the entire group who he works with and to Ultimaker's products and customer support.

SeeMeCNC's Artemis 300 has had the dual extruder option for about a year now and the newer units now come standard with a dual extruder.  In previous versions of Cura the dual extruder was available through configuring instructions that were provided on the SeeMeCNC website. My request was only to see if the dual extruder selection for Artemis could be included with this new update making configuring the printer easier.

Sincerely,

Pete

eman717, 2019-10-09:
I have a customized Rostock MAX v2 with 2 extruders and a single nozzle. I just wanted to pick the "template" and modify it a bit. Is that so much to ask? Is it really "necessary" for the "third-party" to update the options for their printer, rather than Cura allow me to change that option from 1 to 2... sounds like a dumb excuse to me.

Ghostkeeper, 2019-10-09:
We've always held that if you have the technical capabilities to modify the hardware and firmware of your printer intrinsically, you'll also have the technical capability to write a JSON file for it. This allows us to tone down the complexity of our GUI and improves maintainability of the code: not everything has to be able to change on the fly so we can actually cache things. You'll still have an interface, but it's not graphical. It's a JSON file.

We can help you with improving documentation where it's missing, but we can't really write an extruder definition for you without having access to that printer or time to spend. As a template I suggest copying the extruder 0 file and looking at which settings from fdmextruder.def.json you need to override.

nallath, 2019-10-09:
>  His response to my request only reflects negatively on the entire group who he works with and to Ultimaker's products and customer support. 

Perhaps, but on the other hand, you're not a customer of Ultimaker but of SeeMeCNC. His response might have been blunt, but yours was a complete escalation of that (and it doesn't make it any less bad). I filtered your comment because you chose to launch a personal attack on someone. If anyone here would do such a thing, I would filter it. 

> In previous versions of Cura the dual extruder was available through configuring instructions that were provided on the SeeMeCNC website. My request was only to see if the dual extruder selection for Artemis could be included with this new update making configuring the printer easier. 

This is the first time that you've mentioned this. That's okay, but it is where the confusion is coming from. Putting the right files in Cura to support a certain printer is the responsibility of whoever made the printer (or someone from the community that is willing to share profiles).

closed 2020-06-19 by no-response[bot]