Ultimaker/Cura #6494: File folders not showing

pjkoths26, 2019-10-07:
When I select "Open Files" it does not show all file folders that are actually there to be viewed.

nallath, 2019-10-07:
Please use and fill in the bug template as provided. We need that information in order to diagnose the issue.

pjkoths26, 2019-10-07:
I did use the github provided template.

Ghostkeeper, 2019-10-08:
Nothing of that is visible to us or in your edit history, so maybe you accidentally erased it before posting it.

We need to know in particular for this issue:
* Application version. Are you using the latest stable version or an earlier one?
* Operating system. Windows? Linux?
* Reproduction steps. Which buttons did you press? Did you go through File -> Open File(s)? Press Ctrl+O? Or press the button in the main window?
* Actual results. What folders did you see?
* Expected results. What folders didn't you see?
* A log file.
* Maybe some information on whether you have hidden folders visible or not.

The template also lists the printer that you're using but that is probably not relevant to this particular bug.

closed 2020-06-19 by no-response[bot]