Ultimaker/Cura #6495: Different layer height for wall and infill

MichelCHAVOT, 2019-10-07:
Having a different layer height for walls and infill would be a good idea. For example, a layer height for walls equal to half or one-third of the infill layer height (i.e. 2 or 3 layers for the wall and only one for filling) would allow for a good finish and a reduced printing time.

nallath, 2019-10-07:
Correct! That's why there already is a setting for that. Have a look at `Infill Layer Thickness`

closed 2019-10-07 by nallath

MichelCHAVOT, 2019-10-07:
Waouh ! It already exists! bravo! I had not seen this setting. Thank you very much.

Provenance : Courrier pour Windows 10

De : Jaime van Kessel
Envoyé le :lundi 7 octobre 2019 11:12
À : Ultimaker/Cura
Cc : MichelCHAVOT; Author
Objet :Re: [Ultimaker/Cura] Different layer height for wall and infill(#6495)

Correct! That's why there already is a setting for that. Have a look at Infill Layer Thickness