Ultimaker/Cura #6496: Splash screen is slow

BagelOrb, 2019-10-07:
**Application version**
4.3.0

**Platform**
Linux

**Steps**
- Select Cura launcher
- hit enter
- wait for splash screen

**Actual results**
Splash screen takes 10 seconds before it shows up

**Expected results**
Splash screen shows up within first second.

**Additional information**
I frequently open Cura 2 or 3 times because I am unaware that my previous actions to open Cura had succeeded. This is an inconvenience, because loading all those instances of Cura takes long and then closing them down also takes a while.

Ghostkeeper, 2019-10-07:
Which plug-ins do you have installed? These get loaded before the splash screen is created.

Plug-ins need to be loaded before the splash screen is created, because we need to have the user preferences to display the splash screen.

fieldOfView, 2019-10-07:
I am fairly sure there's a "Loading plugins..." stage when the splashscreen is already showing. The splashscreen is shown before all plugins are loaded.

smartavionics, 2019-10-07:
Loading my current Linux AppImage, it takes approx 4 seconds before the splash screen appears and the first status line it shows says "Loading plugins...". This is on a low-end computer (Intel(R) Core(TM) i3-6100U CPU @ 2.30GHz). The Pi4 takes around 7 seconds for the splash to appear.

BagelOrb, 2019-10-07:
I got:

Automatic Slicing toggle Switch 3.5.1 (inactive)
Custom Settings Visibility Set Creator 3.5.0 (inactive)
Mesh Tools 3.5.2 (inactive)
Printer Settings 3.5.2 (inactive)
SVG Toolpath Reader 1.1.1 (active)
Setting Guide 2.0.2 (active)
Sidebar GUI 4.0.4 (inactive)
Titlebar Info 3.5.2 (active)

Oh I just found out the materials and profiles are also listed in the market place. It could be a good idea to have these three categories collapsed s othat users know that the Instlaled tab doesn't only have the plugins listed...

BagelOrb, 2019-10-07:
It would also be nice if all default Cura plugins are not shown automatically. It's hard to find which things I had installed throughout the list...

closed 2024-12-07 by github-actions[bot]