Ultimaker/Cura #6498: monitor:  should stop go beyond the print scope but not  currently

lyfbuaa, 2019-10-07:
monitor:  should stop go beyond the print scope but not  currently

fieldOfView, 2019-10-07:
Please use the issue template, and try to do a better job at describing what the issue is. As is, there is preciously little that can be done about your issue because there's no way for us to read your mind and tell what you actually mean.

lyfbuaa, 2019-10-07:
**Application version**
Ultimaker_Cura-4.3.0.AppImage

**Platform**
fedora 30 x86_64

**Printer**
diy printer like ultimaker

**Reproduction steps**
1.printer settings: set x 200,y 200 z 200
2.montior:print control
  click x+210

**Actual results**
extruder move toward x+210,but not move after reach x 200

**Expected results**
stop at x 200

**Additional information**

rburema, 2019-10-08:
As I understand it; You want the print-head to _not move at all_ when a command is given to travel outside of the build-volume, instead of moving towards that direction and then stopping at the edge.

Is this correct?

Ghostkeeper, 2019-10-08:
Cura doesn't know the current position of the print head, so we can't implement this. In fact, even the firmware doesn't always know the position of the print head unless you home it first.

closed 2019-10-08 by Ghostkeeper

lyfbuaa, 2019-10-08:
add click button for M114 ccommand？
user don't know many gcode command

lyfbuaa, 2019-10-08:
> Cura doesn't know the current position of the print head, so we can't implement this. In fact, even the firmware doesn't always know the position of the print head unless you home it first.

add click button for M114 ccommand？
user don't know many gcode command

Ghostkeeper, 2019-10-08:
That would be a duplicate of your other ticket here: https://github.com/Ultimaker/Cura/issues/6497

I don't think a click button would be appropriate. Just automatically shouldn't be a problem.

However in this case it can't prevent you from going outside the build plate because even the firmware doesn't know the position unless you clicked the home button first.

lyfbuaa, 2019-10-08:
Their scene are different,but can a way  for solving.