Ultimaker/Cura #6500: Printer profile UM2Go with Olsson upgrade has no variants

SmithyAT, 2019-10-07:
**Application Version**
Cura 4.3.0 (but was in earlier versions too)

**Platform**
macOS 10.14.6

**Printer**
Ultimaker 2 Go

**Steps to Reproduce**
Create an Ultimaker 2Go printer and select the Olsson block via "Select upgrades" button

**Actual Results**
With the Olsson block upgrade selected, I got the expected dropdown below the material dropdown (Marlin flavor) to select the nozzle. But only the 0.4 nozzle is shown.

**Expected results**
The dropdown should show the selection (0.25, 0.4, 0.6, 0.8). The same behavior like the UM2+

**Additional Information**
I tried to debug/fix it myself, it seems the variants files are missing. But I had some side effect after creating the files based on the UM2+. So it seems there is more to do or it is just my missing understanding how the definitions, variants and quality profiles work together.

Ghostkeeper, 2019-10-08:
Looks like this has been a problem for a long time. I happened to have 3.6 open and it was there too.

In Cura 4.4 we've "fixed" this problem because we neglected to create an Olsson variant for the UM2Go but if the Olsson block was available for the UM2Go then it should be created there too.

SmithyAT, 2019-10-08:
I like such fixes - keep it simple and maintainable :-)
I guess no much people have an Olsson block in their Go printer and so no one will have noticed.