Ultimaker/Cura #6502: print "two color world"  mistake

lyfbuaa, 2019-10-08:
Application version
Ultimaker_Cura-4.3.0.AppImage

Platform
fedora 30 x86_64

Printer
diy printer like ultimaker with marlin 1.0.2-2

**Reproduction steps**
1.download stl  :https://www.thingiverse.com/thing:11660
2.cura load stl(land and water)
3.cura edit menu->select all models ->merge models
4.slice to print
5.after print layer 1,extruder move toward (0,0）

**Actual results**
after print layer 1,extruder move toward (0,0）

**Expected results**
after print layer 1,continue layer 2,3.....
and normal with Ultimaker_Cura-4.2.1.AppImage

rburema, 2019-10-08:
I can confirm this, also that it doesn't happen in 4.2.1
(Devs: I got a project saved and will attach it to the ticket if we decide to make one.)

Ghostkeeper, 2019-10-08:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5383.

We still have the intention to fix this though. It's a semi-serious issue that doesn't occur often but has undefined behaviour when it does.

closed 2019-10-08 by Ghostkeeper

rburema, 2019-10-08:
Relevant internal ticket: CURA-6279