Ultimaker/Cura #6506: 4.2/4.3 printhead's going home and slooowly back to build plate instead of just start printing

ikkez, 2019-10-08:
**Application version**
4.2.1 and 4.3.0 tested and failed..

**Platform**
Windows 10

**Printer**
FLSUN-QQ Delta Printer
[xyzCalibration_cube_4.0_4.3.zip](https://github.com/Ultimaker/Cura/files/3703667/xyzCalibration_cube_4.0_4.3.zip)

**Reproduction steps**
just slice any model, the bug occurs everytime

**Actual results**
The print head moves down to build plate, extruder starts, but instead moving down to first layer start point, it goes back up and homes again, then going down to build plate again, but reeeeealy slow.. it takes like 10min to go back down again.. I've aborted while waiting.

**Expected results**
The print head moves down to build plate, extruder starts, and then it goes to start point on first layer and start printing

**Additional information**
I went back to Cura 4.0.. the bug is not present there.. my printer profiles where successfully imported to newer versions, I've checked them and just recreated them just to be sure but problem persists.

thank you

Ghostkeeper, 2019-10-10:
Could you provide a project file that reproduces the bug? That is, go to File -> Save... instead of just pressing the button to save the g-code.

ikkez, 2019-10-10:
ahhh okay... my fault. I use Cura since version 3 and there wasnt a predefined FLSUN-QQ in the past when i got the printer, so I ended up with a custom defined printer, with start/end Gcode from FLSUN. I just noticed now that you actually have a matching profile in the current version, so I just tested the default FLSUN-QQ-S printer profile and everything works correctly now. sorry for the false positive and your time spend on this, but heavy thanks 👏

closed 2019-10-10 by ikkez