Ultimaker/Cura #6509: Quite big difference in estimated print time 4.3.0

CZechChuck, 2019-10-09:
Hello,

today I've just finished one print with nice result. But I was quite surprised that Cura estimated time 22hours 20minutes was actually only 18hours 15minutes of real print time. So I'll enclose link for my files if someone would be interested. Have a nice day.

**Application version**
4.3.0

**Printer**
CR-10S

**Reproduction steps**
complete files on attached google drive cloud

**Actual results**
18h15min

**Expected results**
22h20min

**Additional information**
gcode file is compressed due to its size

https://drive.google.com/drive/folders/1KlgBOyflm-QzSCmyrehmVrKOI0ui1W3_?usp=sharing

[image]

Liger0, 2019-10-10:
Update and calibrate your firmware using values that work fine on the machine or set jerk and acceleration in Cura. That's expected behaviour.

Ghostkeeper, 2019-10-10:
Since version 4.2 Cura is not overriding your printer's settings for acceleration and jerk any more. Instead it lets the firmware decide. With various different firmwares going around for the CR-10, it's to be expected that for some firmware versions the acceleration and jerk settings are going to be different from what Cura has stored and so it can't predict well what acceleration and jerk it's going to be using. This causes the discrepancy in the time estimate. There is nothing we can really do anything about, except ask our users to keep the acceleration and jerk up to date in Cura. Sorry!

closed 2019-10-10 by Ghostkeeper

CZechChuck, 2019-10-10:
> Since version 4.2 Cura is not overriding your printer's settings for acceleration and jerk any more. Instead it lets the firmware decide. With various different firmwares going around for the CR-10, [...]

Great explanation, thank you.