Ultimaker/Cura #6510: ironing Issue

kaeltaz, 2019-10-09:
[image]
3d printer after ironing scratches the print surface.

smartavionics, 2019-10-09:
Set combing to "no skin". Also, if you don't use z-hop, perhaps you should?

kaeltaz, 2019-10-09:
I have set combing mode to "no skin". I don't want to use z-hop all over print.

Ghostkeeper, 2019-10-10:
Setting it to No Skin should cause it to retract. Hopefully that reduces the issue. The only other way to prevent scratching the skin is to go over the skin, i.e. z hop. So this is expected behaviour then.

kaeltaz, 2019-10-10:
> Setting it to No Skin should cause it to retract

Unfortunately, this does not help solve the problem.

Ghostkeeper, 2019-10-14:
What is your expected behaviour then?

Ghostkeeper, 2019-10-14:
Moved to https://github.com/Ultimaker/Cura/issues/6517 I guess.

closed 2019-10-14 by Ghostkeeper