Ultimaker/Cura #6511: Material temperatures (build plate, printing) are not being used in Ultimaker 2+ extended

vanerp, 2019-10-09:
**Application version**
4.3.0

**Platform**
MacOs Mojave

**Printer**
Ultimaker 2+ Extended

**Reproduction steps**
*  selected a material with a build plate temperature setting of 50c and a material temperature of 200c
* sliced the STL in Cura
* exported it to an SD
* printed it on the printer

**Actual results**
The printer starts printing with the default temperature of 210c and a build plate temperature of 60c. This is too hot so the print starts warping.

**Expected results**
The printer starts printing with the set temperature of 200c and a build plate temperature of 50c

**Additional information**

rburema, 2019-10-10:
What's the material selected on the printer? The default gcode-flavour used for the UM2 family of printers doesn't include the temperature etc. by design, it instead relies on the material selected on the printer (not in Cura, though those selections should match).

If the right material is indeed selected on the printer itself, a workaround would be to change the gcode flavour (reprap for example is supposed to work too).

vanerp, 2019-10-10:
Thanks for the repsonses

> What's the material selected on the printer? The default gcode-flavour used for the UM2 family of printers doesn't include the temperature etc. by design, it instead relies on the material selected on the printer (not in Cura, though those selections should match).
> 
The material on the printer is PLA. In Cura I use a custom material with slightly different temperatures than the default on the UM2+

> If the right material is indeed selected on the printer itself, a workaround would be to change the gcode flavour (reprap for example is supposed to work too).
I'll try with another flavour of gcode (Marlin and RepRap) and see how that works out.

rburema, 2019-10-10:
> The material on the printer is PLA. 

Then it'll print with the standard PLA temperatures, this is by design.

> I'll try with another flavour of gcode (Marlin and RepRap) and see how that works out.

Great! Then I'll close this. If you still have problems, please feel free to reopen.

closed 2019-10-10 by rburema