Ultimaker/Cura #6512: Cura 4.3 Windows 10 on Microsoft surface pro

Womble62, 2019-10-09:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

New version of Cura will sometimes not install on my Windows 10 Surface pro, when it does I get your configuration is corrupt, then when I eventually get a proper install, it will not always load off the start menu or from a shortcut on the desktop (it does not install a shortcut on the desktop) If I go to change a setting in the top bar as in material details, it does not initiate thm until I have rebooted my pc, STL files no longer show as Cura files, if I try to open an STL file I have to select Cura in the windows box to open it, it will not open from Fusion 360 when I say print, this version just does not seem compatible with my Surface Pro which is the top model....

Womble62, 2019-10-10:
> Could you attach the logs (from when it went wrong)? They should be in %appdata%/cura/4.3 Was there an older version of Cura on your tablet before you installed this one? If so, could you attach the configs in %appdata%/cura/<older version> as well?

[Image]

Hi

I am not sure where to find these things, sorry. I notice on the Web a few people have the same problem.

Thanks

Best regards

Ian Willacy

rburema, 2019-10-11:
Hi @Womble62 

You can just type that into the adress bar of file-explorer (the first one at least, for the second you need to replace `<version>` with the right verison-number). It functions as a handy shortcut to `C:\Users\<username>\AppData\Roaming\cura\<version>` (hwere `AppData` is a hidden folder where a lot of programs, Cura included, save user preferences and the like).

closed 2020-06-19 by no-response[bot]