Ultimaker/Cura #6513: Settings hidden by active machine

3DTanny, 2019-10-10:
**Application version**
4.3.0

**Platform**
Windows 10

**Printer**
Custom FFF printer
Anycubic i3 Mega
Ultimaker S5
Creality Ender 3
Creality Ender 5

**Reproduction steps**
Create new machine profile, enter setting visibility selection

**Actual results**
Several settings are disabled and show "setting hidden by active machine" when the machine profile used to allow the setting visibility. A few that I have used before and can no longer set:
Default Nozzle/Bed Temperature
Retraction minimum travel
Enable Prime Blob

**Expected results**
Really basic settings like retraction minimum travel should not be hard-hidden.

**Additional information**

rburema, 2019-10-10:
The 'retraction minimum travel' _should_ be there, but is currently disabled due to a pending (cluster of) bugs; See [here](https://github.com/Ultimaker/CuraEngine/pull/1108) for an attempt to fix it and ongoing discussion. There's currently a ticket in our backlog (internal ticket CURA-6675) that we haven't managed to get to yet, sadly.

Concerning the 'Default' settings: Those should probably not have been there in the first place, instead of 'Default Printing Temperature' and 'Default Build Plate Temperature' use 'Printing Temperature' and 'Build Plate Temperature' instead.

Enable prime blob was another one that was disabled on purpose. For the UM-S5 (as opposed to the UM-3) in particular it was not deemed necessary, especially since we switched the default behaviour to skirt. The rationale (for the other ones) was that it should be overridden by the printer configuration.

Ghostkeeper, 2019-10-10:
> The rationale (for the other ones) was that it should be overridden by the printer configuration.

The rationale for the other printers is that this feature works via the `G280` g-code command which is not by default in mainline Marlin but just something that Ultimaker's firmware engineers decided on.

ctmakro, 2019-10-16:
confirm: setting 'Retraction minimum travel' 'hidden by active machine'.

zmp2000, 2019-11-07:
So following the [6675](https://github.com/Ultimaker/CuraEngine/pull/1108) it has been fixed but not merged yet? I installed the 4.4 Beta but still not fixed yet.

Ghostkeeper, 2019-11-07:
It's not been fixed yet, even on that branch. The issue was more difficult than we initially thought so we've moved it back to our backlog.

jlgill, 2019-12-24:
Is this a bug for specific printers or is this Cura-wide? I ask because up until today I was able to change this setting without issue, but for some reason all of sudden the setting is now hidden and I can't change it.  

Googling brought me here but I was actually able to change this setting once and then it was hidden by the bug.  Does changing the setting actually apply?  I have a Anycubic Chiron that was having some pretty bad stringing on smaller prints and I changed this setting from 8mm to 1.5mm and it appeared to drastically improve the stringing issue but now I can't change it anymore due to this bug.

Ghostkeeper, 2019-12-24:
Changing the setting would have no effect. But this has been fixed in the next version. You'll be able to change it again then.

closed 2019-12-24 by Ghostkeeper

bleker3d, 2021-02-28:
Hi,

I have this issue again. I can not select the ooze retract distance or speed ? There are a lot of my settings blocked - "This setting has been hidden by the active machine and will not be visible"

On the first install of Cura 4.8 I selected my printer a Tronxy x5sa Pro and it did not have the settings active. I uninstalled Cura completely and reinstalled selecting custom machine, same problem.

[image]

rburema, 2021-03-02:
@bleker3d Those settings are defined/configurable only per material or per printer (so if you where to manually make profiles for those). Some others may be hidden because they only make sense for a dual(+) extruder machine, etc.

Something like this would usually be set by a material or machine manufacturer only, and are not in general meant to be tinkered with per print.

Some of the ones hidden for being a material setting you can adapt by cloning the material in material settings, whereupon the text-fields are enabled (like 'Default Build Plate Temperature' (but you can just set the 'normal' Build Plate Temperature, which overrides the former, so there is generally no need to do that, unless you want to define you own material profile).

As for the rest, there _is_ a plugin that allows changing those. While in general, Cura is already use at your own risk, this plugin is that turned up to 11. Please be careful, there's a reason these settings where hidden: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PrinterSettingsPlugin

(The original bug in this issue was that they found settings that _shouldn't_ have been hidden, but where. I see no such settings labeled with ℹ️ in the screenshot you provided.)

Ghostkeeper, 2021-03-08:
We hide those material settings, by the way, because they are not used by Cura to generate g-code. They needed to be in the material profiles because Ultimaker's firmware uses them to change filament.

wickedwingsrc, 2021-03-16:
hey guys had same problem, settings hidden. It is BS, seems learning this is a challenge!

Here is the fix, it does work, because I could not get the material temp settings, I did what the video says, so simple, and typed Temperature in the search in the settings menu and it popped up. I had to tinker to get all the settings to show again but now I can change temp again.

https://youtu.be/v7Dle6lB4VM

3DTanny, 2021-03-17:
My solution was to switch to Prusa Slicer because it is a much better
Slicer. 😘

loshlee, 2021-04-12:
I love the config, thanks. I have one problem. Prime Blob settings under Build Plate Adhesion (•Enable Prime Blob, •Extruder Prime X Position, •Extruder Prime Y Position) are disabled. Would you enable them, please?