Ultimaker/Cura #6516: GUI elements missing

Evilspawn-USMC, 2019-10-10:
**Application version**
4.3.0

**Platform**
Windows 10; Version: 1903; Build: 18362.418
NVIDIA Quadro K2100M; Driver version: 23.21.13.8908

**Printer**
Ultimaker 2

**Reproduction steps**
This is my work computer. When the sysadmins pushed an update last night I'm no longer able to see most of the GUI in Cura. I've attached a screenshot for your reference.

**Actual results**
See Screenshot

**Expected results**
A usable GUI

**Additional information**
[image]

rburema, 2019-10-11:
Hm, sounds like this users' [comment on another thread](https://github.com/Ultimaker/Cura/issues/6387#issuecomment-535646459), which was somehow solved by reinstalling 4.3 (without deleting the configs).

In any case; Is Cura still actually running on the nvidia card or is it (after the update) somehow picking up an on-board intel card? Do you have a dual monitor setup? Could you send us the logs?

Evilspawn-USMC, 2019-10-11:
I could, but I'm not exactly sure how to do that. If you can give me some guidance on what you'd need and how to get it, I'd be happy to.
I did force it to run on the NVIDIA card and it still isn't working.

Ghostkeeper, 2019-10-14:
Sounds like this issue to me as well: https://github.com/Ultimaker/Cura/issues/6003

Could you try updating your video drivers?

Ghostkeeper, 2019-10-14:
> I could, but I'm not exactly sure how to do that. If you can give me some guidance on what you'd need and how to get it, I'd be happy to.

Instructions on how to get the log files are here: https://github.com/Ultimaker/Cura#logging-issues

closed 2020-07-06 by no-response[bot]