Ultimaker/Cura #6517: ironing Issue #2

kaeltaz, 2019-10-10:
[image]
3d printer after ironing scratches the print surface.
I have set combing mode to "no skin". I don't want to use z-hop all over print.

alexchiolo, 2019-10-11:
This feels like a feature request (which I would second) - a toggle to enable z hop on ironed layers.

3dprintrookie, 2019-11-15:
Would love this feature... I'm getting the same scar marks from ironing on 4.3

Ghostkeeper, 2019-12-03:
What would you expect the printer to do, when it needs to travel from one place to another across the skin? You say that you don't want to use z-hop all over the print.

alexchiolo, 2019-12-03:
This could be implemented as a toggle "Z Hop Only Over Ironed Parts" similar to "Z Hop Only Over Printed parts", than only appears when Ironing is enabled. Basically add perform a Z hop when crossing the skin, but only during the ironing pass. This should have a negligible impact on print speed and Z axis wear, while significantly reducing scarring.

Ghostkeeper, 2019-12-17:
While it's nice to have, we needed to defer this since we don't have time for it now. We'll defer this issue, so don't expect it to be fixed soon.

SilverWx, 2020-03-26:
I have this problem too, please add this option :D Thanks you for your great work in cura 4.5, that being said.

A-Mueller30, 2021-01-15:
Are there any updates on this feature? I love the ironing feature, but the z-hop would make it awesome :)

ltjessem, 2021-04-16:
Bumping this as it's ruining otherwise good ironing

dawithers, 2021-06-06:
Tried ironing for the first time today and am baffled by this. Lack of this feature totally defeats the purpose of ironing.

Eddie-Hartman, 2021-12-30:
@Ghostkeeper If you'd prefer me to open a new issue, let me know, but I wanted to place this here since it's related to this thread.

After hours of research and trying to understand these various settings, I'm reaching out for help because I believe I've ran into a bug. If I have a setting clearly wrong, please let me know.

When I have ironing on with combing set to "Within Infill" and "Z Hop When Retracted" turned on, I'm getting scarring on my surface and looking at the gcode, it doesn't appear that it is z-hopping when it should. I was fine with using z-hopping everywhere in my print since it is small as you mentioned [here,](https://github.com/Ultimaker/Cura/issues/6517#issuecomment-561254471) but that's kind of why I'm stuck and wondering if there is something I'm missing or if this is a bug.

Screenshot of what I mean:
[image]

Project File:
[Mayvue CoinGithub.zip](https://github.com/Ultimaker/Cura/files/7791285/Mayvue.CoinGithub.zip)

On layer 12 around the dots you can see the scarring. It's actually happening all over the place.

After a whole lot of experimenting, I think I've isolated the problem to this setting: "z hop only over printed parts".

For some reason, toggling this on or off is resulting in different travel paths. When the setting is on, it's not z-hopping like I would expect, which is causing the scarring line above. With it off, the gcode generated appears to not cause scarring. (I can actually do a print soon, but this is just judging from the resulting gcode. As an aside, having to actually save the gcode then open the gcode to view z-hopping lines is such an unexpected user experience. Is that getting addressed at all?)

My expectation was that since this is traveling over a printed (and at this point ironed) part that it would z-hop to avoid scarring.

Some quick googling shows that other users may be experiencing this issue as well:
- https://www.reddit.com/r/FixMyPrint/comments/lx5n5j/so_in_cura_i_have_z_hop_only_over_printed_parts/
- https://community.ultimaker.com/topic/35796-issue-with-no-z-hop-over-printed-part-even-with-proper-settings/

I'm fairly new to the Z Hop and Ironing settings and all of this came from hours of trial and error viewing gcode. I attached the project file to hopefully help debug, but there is still a good chance this is just simply user error. Thanks.

Ghostkeeper, 2022-01-11:
It's best to post that as a separate bug, indeed. It has nothing to do with specialising Z hop to only hop over ironed parts, really. Not all Z hop issues should get posted here.

I think your issue is that Z Hop Only Over Printed Parts only causes it to hop over parts that are already printed (as per the expected behaviour of the setting) and that doesn't include the part that is currently in the process of printing.

jstarrfire, 2022-02-25:
I've been attempting to work around this for several days, but have had zero success. Unfortunately without z hop, there are always major scars - and ironing is essentially useless.

LMBernardo, 2022-04-02:
> This feels like a feature request (which I would second) - a toggle to enable z hop on ironed layers.

It's not really a feature, it's a bug with ironing. Every time I use it, everything goes perfectly until the end. Then it gouges right across the surface it just ironed, completely defeating the purpose. I definitely understand that there are bigger priorities but I don't think it's right to treat this as a feature request, it's an issue with the implementation of an existing feature.

den-mac, 2022-05-24:
Bump - I agree with everyone I watch the ironing process and it's smooth as silk, but then the nozzle moves to the next part and leaves a scar. A simple z-hop would alleviate this problem and since it's usually the most visible surfaces we iron having this simple feature to ensure a near perfect product seems like an important feature.

MonikaLewandowska1991, 2024-03-17:
Maybe finish Ironing on the outer wall (edge) then dragging will not occur?
Or just add Z hop after Ironing (this could be achieved with Before layer change g-code.