Ultimaker/Cura #6519: [4.3.0] Acceleration and Jerk settings seem to have no effect for Ender 3

aethersis, 2019-10-10:
**Application version**
4.3.0

**Platform**
Windows 10 Professional x64 (up to date)

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
Try to slice any stl file with default settings, increase jerk and acceleration

**Actual results**
Print time does not decrease

**Expected results**
Print time should decrease. For the same settings 4.3.0 is slower by up to 40% in some cases than Cura 3.6.0

**Additional information**
Could it be because Creawesome mod is now enabled by default and adds gcode to control acceleration and jerk at the beginning? Is there a way to permanently disable it / root this nasty modification out? If not then I believe there should be an option for this.

rburema, 2019-10-11:
Keep in mind that for 3rd party profiles we're (basically) wholly dependent on the community.

> Could it be because Creawesome mod is now enabled by default ...

We did pull in those profiles.

> ... and adds gcode to control acceleration and jerk at the beginning?

I did have a look at the start gcode and it does indeed look like it does that:

~~~
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
<... snip rest ...>
~~~

The most straightforward workaround is to open up the machine settings and alter the start gcode there. There's even a little pre-processing there so you can put values between brackets like for example `M104 S{material_print_temperature}`

Another workaround would be to either modify the current profiles, or create your own profiles (starting perhaps, from an older version of the base file).

Ghostkeeper, 2019-10-14:
Cura assumes that the maximum acceleration and maximum jerk settings are equal to the settings defined in the printer definition. Assuming you're using Ender 3 instead of Ender 3 Pro, that would be these settings:

https://github.com/Ultimaker/Cura/blob/4cb59b54545c4e4894d96a73cf85f13b83f02203/resources/definitions/creality_base.def.json#L130-L143

If you want to change the maximum acceleration and feedrate, you'd also have to change them there (or load in the Printer Settings plug-in from the Marketplace to change them in Cura's GUI).

Or if you want to change the normal acceleration and jerk settings, check the Enable Jerk Control and Enable Acceleration Control settings. That's the intended use for those settings. Cura won't execute your start g-code to see what the acceleration/jerk ends up at.

aethersis, 2019-11-10:
Thank you, Ghostkeeper, that solves my issue I think!

rburema, 2019-11-12:
> or to disable the mod so that people know what's going on.

The 'mod' was mostly just a lot of profiles, that where different from the ones we already had. We've since overwritten these profiles, so there isn't really a way to 'disable' the mod.

> Still, I think it would be useful to add some more obvious setting for those limits

Limits that are not really normal printing settings, but should really be already set correctly for each individual printer. I know in cases like these, where something changed (lesson where learned during the fallout...) it can be annoying, but I think the plugin adequately addresses those complaints.

Since the issue is otherwise solved, I'm closing this issue.

closed 2019-11-12 by rburema