Ultimaker/Cura #6520: Feature Request: Don't support bridges

johnkray, 2019-10-10:
The PrusaSlicer has a very useful feature called "Don't support bridges," which when enabled will not generate support material under areas of the part that are using bridging. This would be a great addition to Cura as it saves print time, material usage, and post-processing time.

Other than a checkbox to enable or disable the feature, I think it would also be useful to be able to define the maximum unsupported bridge area. This way supports for small bridges could be disabled, but support material for large bridges could still be generated.

MastaRob, 2019-10-13:
+1 - coming from Repetier this is a very useful feature. 
Also I cannot seem to find a setting for maximum support height, i.e. no support above a certain z distance. This is another feature I used all the time in Repetier, and must be easy to program in.

Ghostkeeper, 2019-10-15:
We've previously made an attempt at a maximum unsupported bridge area. This setting is called Minimum Support Area. Due to the way it works, it tends to remove small parts of support that should probably not be removed as well.

To really fix this we'd need to take the bridged areas instead and do something with those, I think. That would make it much more focussed and therefore much more effective. And it should also probably remove areas of support with a Minkowski subtract using the directions of the line (intersect the shape with a shifted version of itself or something).

However we won't have time to implement this soon, so we're deferring the implementation of this. We'll keep it open until we get some more time.

elixirtrip, 2019-12-24:
I found out, that using support x/y distance and support distance priority on z, and minimum support x/y distance may solve some cases, not all but something...

EmreSURK, 2022-09-05:
This is a definitely underestimated feature. 
When you are printing a complex modal, you need to add a lot of configs to run away from a lot of support by using plugins. Printers are capable of printing bridges without support but the lines that start in the air definitely need.

andrewaylett, 2023-09-05:
I would appreciate this feature too -- thank you for considering it.

anubis71x, 2024-01-26:
I agree that this is a _small_ issue. But it would be nice.

NotThePassword, 2024-03-13:
I am chiming in, hoping someone sees it. This would save a lot of material and/or time spent trying to configure 8 settings and different support blockers to print certain files.

creativspelerr, 2025-04-13:
Instead of quantifying a bridge based on the amount of area, it should be based on the span of the bridge.  If I were printing a long inverted U channel the area of the unsupported bridge wouldn't matter, the fact that all the bridging is short would make the print successful with no supports - but there might be need for supports on some other area.