Ultimaker/Cura #6521: [4.3.0] Initial Layer Flow is not applied as an absolute value

alexchiolo, 2019-10-10:
**Application version**
4.3.0

**Platform**
Windows

**Printer**
Creality Ender-3

**Reproduction steps**
Starting from the default standard quality profile I changed the first layer flow to 105 (to optimize first layer on PETG), and a test square printed perfectly, then in a subsequent print I changed the main flow value to 90 to tune vertical wall thickness and it unexpectedly affected the first layer as well.

**Actual results**
The inside surface of the bottom layer is underextruded, the 90 value is applied as well as 105, resulting in a 94.5% flow.

**Expected results**
Initial layer flow set according to the configured value for it.

**Additional information**
Upon further investigation the issue can be reproduced easily in the slicer preview by exacerbating the values, the attached screenshots should be self-explanatory, including a workaround that allows to obtain the expected flow in the first layer (input value=100*{"first layer flow"/"flow")

If the observed behaviour of the initial layer flow being a multiplier for the other flow values is expected, it should be documented properly (both in the tooltip for flow, which doesn't mention initial layer flow as affected parameter, and in the one for initial layer flow mentioning that the extruded material for layer 0 is multiplied by this value AND the relevant wall/bottom values), if otherwise the initial layer flow is meant to be an independent value the calculations need to be fixed.

[image]
[image]
[image]

nallath, 2019-10-11:
Nice find (and thanks for the detailed write up!). I've created an internal ticket for this; CURA-6885

soxhi8, 2020-12-28:
Issue is still present in Cura 4.8.0.

Ghostkeeper, 2020-12-28:
Indeed, looking at that internal ticket mentioned above it seems that our PO has deferred the issue since he thinks we won't have time for it any time soon. It's not fixed because there were other issues that were deemed more important.

soxhi8, 2023-05-30:
Still an issue in Cura 4.13.1.

MariMakes, 2023-05-30:
A fresh set of eyes reporting for duty 🐣 

We updated the initial flow settings in Cura 5.2. 
You can now define the following settings to fit your needs. 
- Initial Layer Outer Wall Flow
- Initial Layer Inner Wall Flow
- Initial Layer Bottom Flow

But I would expect that in this case, the Initial Layer Flow would override the Top/Bottom flow. 
[image]

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

MariMakes, 2023-08-29:
Quick update from our side 👋

I've discussed this with the team and they find it confusing as well. 😅 
I will mark this as a duplicate of https://github.com/Ultimaker/Cura/issues/14874
We have a ticket on our backlog with the intent to improve this behavior, but this behavior has existed for quite a while, so we'll have to analyze if the risk is worth the impact. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14874

Apologies for the wait, hopefully, we can resolve this to make it easier for everyone.

closed 2023-08-29 by MariMakes