Ultimaker/Cura #6522: On bottom layer hole perimeter does not connect with body of print

bill-orange, 2019-10-11:
Application version: 4.3

Platform: Windows 10

Printer: Anet A8 fed by Octopi running on Raspberry PI B+

Reproduction steps: virtually any hole on first layer shows the disconnection if top/bottom set to lines (see photos)

Actual results: perimeter of holes not connected to body of first layer

Expected results: continuous first layers

Additional information:

I have developed a new problem that I am at a loss to explain.

Most of the time I am printing small project boxes for micro-controllers. I have usually gotten good surface finishes on the lid. Recently I have started modeling the boxes in Fusion360 rather than Tinkercad.

Here's the problem. The lids normally have rectangular holes for OLED displays or other stuff. I have started getting odd unattractive gaps between the holes and the body of the lid on the first layers. I watched the printer print the first layers today and saw what you see in the photos. No wonder the appearance sucked! I then looked at the GCODE viewer in Octopi and saw what you see below. You can see head lifts where I see holes in the print.

What's wrong and how do I fix it? Is this a Fusion 360 problem, a Cura 4.3 problem, a printer problem or operator error?  I strongly suspect Cura
[image]
[image]

[LID_TWO_HOLES.zip](https://github.com/Ultimaker/Cura/files/3715070/LID_TWO_HOLES.zip)

bill-orange, 2019-10-11:
I am certainly open to the idea that it is a printer issue, however I consider it unlikely since the problem is symmetrical about both the X and Y axis.  A problem on both axis resuliting in a symmetrical issue seems unlikely.  In any case bot belts are tight on the Anet A8 (30 Hz twang).  

I am more inclined to suspect this is a manifestation of the problem reported in #6419 .  Perhaps the same underlying problem causes both conditions.

smartavionics, 2019-10-13:
Hello @bill-orange , could you please zip up the gcode file that produced the print shown above and attach to this issue. Thanks.

bill-orange, 2019-10-13:
@smartavionics   I _think_ this is it.  I had already deleted the offending gcode and had to retrieve it with Windows 10 File History.  Its difficult using file history to tell just what you are recovering sometimes.  Well, In any case I ran the code and it exhibits the symptoms.
[Normal_LID_TWO_HOLES.zip](https://github.com/Ultimaker/Cura/files/3721878/Normal_LID_TWO_HOLES.zip)

smartavionics, 2019-10-13:
Thanks for the gcode. I have a few suggestions to improve the chance of the the skin lines sticking to the walls around the hole.

1 - set extra skin wall count to 0.

2 - set skin overlap percentage to at least 10 or 20.

3 - print the first layer walls and skin slow!

4 - maybe try zigzag pattern instead of lines.

BTW, the gcode is not lifting the nozzle between the ends of the skin lines, it's just doing a travel move.

Ghostkeeper, 2019-10-14:
> 2 - set skin overlap percentage to at least 10 or 20.

I think this is a very good suggestion. The skin overlap setting was made to combat this particular issue. One of the effects is that when the line stops exactly at the wall and the nozzle turns around, the line gets dragged along back with the nozzle before it gets a chance to attach to the wall. This produces a gap.

> 4 - maybe try zigzag pattern instead of lines.

This will probably magnify the problem since the zigzag pattern stops a line half a line width before the wall instead of on the wall, to prevent overextrusion. If you try this, it's wise to increase the skin overlap percentage even more (50% more or so).

bill-orange, 2019-10-14:
@Ghostkeeper  In notice you added the tag 'Status:Needs Info'.

What additional info is needed?

Ghostkeeper, 2019-10-15:
Sorry for forgetting to mention that. I'm wondering if adjusting the skin overlap solves your problem.

bill-orange, 2019-10-15:
Yes,, skin overlap set to 18 solves the problem.  Perhaps this should be the default.  Is there a downside in some situations?
[image]

closed 2019-10-15 by bill-orange

Ghostkeeper, 2019-10-17:
Increasing it too much causes overextrusion which leads to blobs. I've written an article about the setting in my Settings Guide plug-in here: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/skin_overlap.md