Ultimaker/Cura #6523: extruder output at different layerheights

maukcc, 2019-10-11:
We were testing extruder output at different layerheights and found strange conclusions.

When a specific nozzle size does twice the layerheight (0.4mm nozzle doing 0.1mm then 0.2mm layerheight) one would expect to see 200% extruder output.
But the extruder output is 240% for any nozzle.
We only tested this for 1st layer.

Is this bug or feature?

Also when doing 0.4mm nozzle compared to 0.8mm nozzle for the same layerheight the output for the 0.8 is 200% of the 0.4 , so that seems correct.

rburema, 2019-10-11:
Please be aware that there are several settings that influence this that specifically target the first layer. What does, for example, your 'Inital Layer Line Width' say?

maukcc, 2019-10-11:
As far as we can see everything else stays the same.

Ghostkeeper, 2019-10-14:
I loaded in a 1cm cube and disabled the brim. Here is a snippet of g-code of the very first line when the initial layer height is set to 0.1mm. Note that it starts with the innermost wall:

```
;MESH:cube.stl
G0 F4285.7 X112.46 Y111.54 Z2.1
M204 S500
M205 X5 Y5
;TYPE:WALL-INNER
G1 F600 Z0.1
G1 F1500 E0
G1 F1200 X112.46 Y103.46 E0.0456
```

And here is the same snippet when the initial layer height is set to 0.2mm:

```
G1 F600 Z2.2
;MESH:cube.stl
G0 F4285.7 X112.46 Y111.54 Z2.2
M204 S500
M205 X5 Y5
;TYPE:WALL-INNER
G1 F600 Z0.2
G1 F1500 E0
G1 F1200 X112.46 Y103.46 E0.09119
```

The coordinates of the line are the same: From `(X112.46, Y111.54)` to `(X112.46, Y103.46)`. The extrusion went up from `0.0456` to `0.09119`. `0.09119 / 0.0456 = 1.99978`. So there seems to be some rounding involved which results in 0.02% underextrusion when the layer height is set to 0.2mm (or overextrusion with 0.1mm). But for the rest it seems pretty ordinary to me.

Are you measuring this in the g-code or in the actual nozzle output? There could be some weirdness going on with slip or something in your feeder.