Ultimaker/Cura #6525: Profile name at top

FlailAway, 2019-10-11:
**Is your feature request related to a problem? Please describe.**
Nope

**Describe the solution you'd like**
Put the Profile name at the top of the GCode file somewhere near  the 
";Generated with Cura_SteamEngine 4.3.0-BETA"

I know it is at the end under nname, but it would be nice to see it at the top. I am printing some parts and the Initial stuff has come out prefect. I have changed Profiles in the current running Cura so I have no idea which of the 7 Profiles I have modified for PLA without having to scroll all the way through to the end and visually search in the wall of text there.

Or, better yet a "Last file used" list within Cura showing the Part Name and the Profile.

Ghostkeeper, 2019-10-14:
Some people keep track of their print history by saving their workspace at every print. We didn't implement this though because that could fill up your hard drive quickly.

I think scrolling down to the bottom of the g-code is not really much of a problem. I'll bring it up though with my colleagues to see what they think about it.

Ghostkeeper, 2019-10-15:
Sorry, we're keeping it at the bottom. Just use Ctrl+End or something to scroll down. It's not really an issue and putting it at the top has readability issues too.

closed 2019-10-15 by Ghostkeeper