Ultimaker/Cura #6526: [4.2.1 and 4.3] Raft and Fan Causing Printer to Freeze

RedRightArm, 2019-10-12:
**Application Version**
4.21 and 4.3

**Platform**
Mac OS Mojave

**Printer**
FLSUN QQ Delta Printer

**Steps to Reproduce**
I have the fan set to Fan Speed: 100%, Initial Fan Speed: 0%, Regular Fan Speed at Height: 0mm, Regular Fan Speed at Layer: 1. I have the raft enabled. The raft settings do not matter. This issue occurs occurs due to the fans turning on after the raft is finished printing.

**Actual Results**
When the raft is finished the fans turns on. With a raft used, the first layer it prints is labeled -6 in the g-code. the next layer is -5, the -4, and so on. The last layer of the raft is -1. The first layer of the printed model is 0, but the 2nd layer of the model is Layer 1 in the g-code. Because Cura won't accept a Regular Fan Speed at Layer: 0 value, this makes the fans turn on on the model's 2nd layer. It is on this 2nd layer (Layer 1 in g-code) that the printer freezes. For some reason having the g-code turn on the fans at Layer 1, after the raft has been printed, causes the printer to simply freeze. No error message. No problem noises. It just freezes like pausing a video frame. And it only does this on the 2nd layer (Layer 1 in g-code) when using a raft.

**Expected results**
The raft should print with the fans off. The fans should turn on at the first model layer, stay on for all model layers, and the print should finish as expected.

**Additional Information**
I have tested printing the same model with skirts and brims and they complete fine. I then tested turning all fan settings off and the print is able to complete with a raft. But as soon as I allow the fans to turn on the printer fails on the 2nd layer.) I have also tried this in both Cura 4.2.1 and 4.3 and the issue happens with both versions.

smartavionics, 2019-10-13:
Hi, could you please save the project file (File -> Save) and then either rename or zip the .3mf file to .zip and attach to this issue. Thanks.

RedRightArm, 2019-10-13:
Hello,
While continuing to pick away at this issue I am beginning to suspect that this is a hardware issue. I think there is something going wrong with my fans, motherboard, or power supply. I need to do more trials before i figure it out. With this new gut feeling in kind, would you still like me to send you the project file?

Best regards,
Andrew Newman

smartavionics, 2019-10-13:
No, as it sounds like it's a hardware problem the project file probably won't be that useful. Please close this issue, you can always reopen it again later if you wish.

Ghostkeeper, 2019-10-14:
I agree that from the description there seems to be a problem in the firmware.

Unless there is a pause command in the g-code somehow (`M0`?), Cura seems to be doing its job okay, from your description.

closed 2019-10-14 by Ghostkeeper