Ultimaker/Cura #6527: Cura 4.3. Bed moves in opposite Y direction when printing is finished

pincorr, 2019-10-12:
Hi,
moving from version 3.6 to 4.3 I faced the following problem on my CR10:
in leatest version ,at the end of the print, the bed moves to Y=0 where there is no end stopper with the result that the bed hit the frame. In older version the bed moved correctly to the opposite position where micro switch is. I didn't make any change in Gcodes, and I used the same settings in both software. 

Possible some Gcode line that must be manually adjusted ?

Any advice about how to solve the problem  is appreciated.
Thanks

Ghostkeeper, 2019-10-14:
I'm not seeing anything in the end g-code that could be doing this. Here is the g-code at the end of a print on the CR-10:

```
G0 F9000 X153.151 Y154.19
;TIME_ELAPSED:595.398537
G1 F1500 E262.09336
M140 S0
M107
G91 ;Relative positionning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y300 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

M82 ;absolute extrusion mode
M104 S0
;End of Gcode
```

I see it making some relative movements, including a 5mm movement towards positive Y. But that shouldn't look like the bed is homing and especially not towards Y=0. Could it be a problem in the firmware that happens after the g-code supplied by Cura is executed?

vbz89, 2019-10-29:
The bed is moving forward because of this line:
G1 X0 Y300; Present Print.

If you want the bed to go back to it's home position, you can replace "Y300" with "Y0".

I noticed the same thing with my ender 3 when I printed using stock profile in Cura 4.3. Hope this helps.

closed 2020-06-19 by no-response[bot]