Ultimaker/Cura #6528: Support Interface Roof with Cura [4.3.0]

justinqmalang, 2019-10-13:
**Application Version**

4.3.0
**Platform**

Windows 10
**Printer**

Creality CR-10S Pro
**Steps to Reproduce**

1. Add a model such as a mini that will require advanced supports
2, Turn on Support Interface and put the density as 80%
3. Slice

**Actual Results**

The first few layers of the object and the last layers of the support interface roof layers are joined together as a single layer.
**Expected results**

The layers of the object printing directly on top of the last support interface roof layer.
**Additional Information**

Ghostkeeper, 2019-10-14:
I'm not seeing the problem you describe. Here is a picture of my D&D character's mini:

[image]

The layers are not being joined together and the layers of the object are printed directly on top of the interface. The expected results are met in this case. Do you have an example of what's going wrong? Perhaps you could share a project file?

Are the layers merging together such that there is overextrusion?

justinqmalang, 2019-10-14:
From: Ghostkeeper
Sent: October 14, 2019 07:32
To: Ultimaker/Cura
Cc: justinqmalang; Author
Subject: Re: [Ultimaker/Cura] Support Interface Roof with Cura [4.3.0] (#6528)

I'm not seeing the problem you describe. Here is a picture of my D&D character's mini:

The layers are not being joined together and the layers of the object are printed directly on top of the interface. The expected results are met in this case. Do you have an example of what's going wrong? Perhaps you could share a project file?
Are the layers merging together such that there is overextrusion?

Ghostkeeper, 2019-10-14:
I think something went wrong with your reply to this ticket, sorry! It is an empty message.

justinqmalang, 2019-10-14:
Is there another email I can reach you at? I will try again with this email but in case it doesn’t work, please send me an email I can send the file and screenshot to.

From: Ghostkeeper
Sent: October 14, 2019 10:57
To: Ultimaker/Cura
Cc: justinqmalang; Author
Subject: Re: [Ultimaker/Cura] Support Interface Roof with Cura [4.3.0] (#6528)

I think something went wrong with your reply to this ticket, sorry! It is an empty message.

justinqmalang, 2019-10-14:
[image]

justinqmalang, 2019-10-14:
https://gofile.io/?c=WoEe9E

Here's a link where you can download the project file and also the image of the issue.

Ghostkeeper, 2019-10-15:
Ah, thank you. (By the way, I see in my notifications that you've had trouble with uploading files. Github accepts attaching files as long as you zip them first.) The issue reproduces with that 3MF file.

Looks like this is a bug caused by the Minimum Support Area, and in your case the Minimum Support Interface Area. Setting both the Minimum Support Area and the Minimum Support Interface Area to 0 solves the issue and should be a fine workaround for this print.

The bug was already reported here: https://github.com/Ultimaker/Cura/issues/6147
So I'll close this as a duplicate.

closed 2019-10-15 by Ghostkeeper