Ultimaker/Cura #6529: Position with Cura 4.3.0

GilbertGC, 2019-10-13:
After preparing the object, I save it as usual. 
when I transfer to my 3D printer Anet A8, it do not start at the middle of the bed but just in the left front side of it;
I changed the configuration of the printer in Cura but still the same problem.

GilbertGC, 2019-10-14:
> What did you change?
> In Machine Settings, make sure the "Origin at center" setting is *not*
> checked. This option should basically only be used for delta-style printers.

I change the size of bed
The origin is at center, I check the G code, from Cura,  With another
software and I find the object near the bed!!
There is some problem with Cura 4.3.0 I was using another version
without any problem.

Best regards

Gilbert Couic

couicg@orange.fr

Tel mobile +33(0)683751383

fieldOfView, 2019-10-14:
Like I said above:

> In Machine Settings, make sure the "Origin at center" setting is **not** checked. This option should basically only be used for delta-style printers.

GilbertGC, 2019-10-14:
Ok 
I will try!

GilbertGC, 2019-10-14:
Nice!!
Working correctly now.
the other problem I have also is to connect my printer to Cura, 
Cura did not find it. It is on com3.
and Cura did not see anything.

Ghostkeeper, 2019-10-14:
Using Printrun or something similar, do you know what baud rate the printer is on and how the printer responds to a temperature probe with `M105`?

I'll close this issue since the original problem has been fixed. We can't change it in the mainline Cura distribution since we don't have an Anet A8 printer definition.

closed 2019-10-14 by Ghostkeeper

GilbertGC, 2019-10-14:
ok

Thanks
Best regards

Gilbert Couic

couicg@orange.fr

Tel mobile +33(0)683751383