Ultimaker/Cura #6530: Multi-monitor issue with Splash Screen and Main Window

jay-packet, 2019-10-14:
I've verified an issue. If Cura's splash/load screen appears on my main monitor ( I have three monitors active) and Cura's interface loads on a different monitor it will freeze. As soon as I disconnected the other two monitors to force Cura to load the splash screen and the main screen on the same monitor it fixed the issue. No problems loading and no freezing. I'm using Cura 4.3.0

**Application version**
Cura 4.3.0

**Platform**
Windows 10 Home version 1903 build 18362.418

**Printer**
ANET A8 (printer not running or connected)

Nvida Drivers 436.48 (release date 10/01/2019)
Geforce 980

fieldOfView, 2019-10-14:
This PR work solve this issue: https://github.com/Ultimaker/Cura/pull/6385

Ghostkeeper, 2019-10-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/4894.

closed 2019-10-14 by Ghostkeeper