Ultimaker/Cura #6534: Prime tower

maaspt, 2019-10-14:
Prime tower prints first layer before changing extruder.

When extruder is changed, it does not print prime tower, that layer of prime tower was already printed by the extruder that was printed first.

Prime tower should be printed only after extruder change, and color change.

Ghostkeeper, 2019-10-15:
Cura considers it more important to print the first layer of the prime tower with the same material as the outer shell of the prime tower. The outer shell is always complete from bottom to top and is used for strength. Different materials don't adhere as well to each other as otherwise, so if the first layer is printed with the other extruder, the prime tower has more chance of falling over which ruins the print.

In the first layer, the skirt or brim is supposed to be priming for you. Is it not priming?

maaspt, 2019-10-15:
No, in first layer it primes extruder 1 before printing part, and after skirt.

I removed brim or skirt to test, changed many setting, and getting always the same result.

I think this started after updating to Cura version 4.0, before was working ok.

The problem is that, in first layer with two colors, after printing prime and part with Extruder 1, it changes to extruder 2 but it does not prime after that change. Second color gets printed with remaing color from extruder 1.

Extruder 1 primed before part

[image]

Extruder 1 printing

[image]

Extruder 2 printing without prime

[image]

Second layer extruder 1 not priming ok

[image]

Ghostkeeper, 2019-10-17:
If you disable the skirt and brim then I'd indeed expect that it doesn't prime. The same will happen when you disable the skirt and brim in single extrusion prints.

If platform adhesion is enabled I'd expect it to prime an extra line with all extruders though. I've made a screenshot of that here when using brim:
[image]

And here from my settings guide when using skirt:
[image]

The function of the skirt is to prime. You can disable it if you don't want it to prime unnecessarily (e.g. if your printer's firmware always primes before a print or something). But normally you'll need at least a skirt.

AbeFM, 2019-11-05:
I've noticed the same thing - it doesn't really effect my prints - but one COULD save a little material by having the skirt be only in the second material, allowing you to prime that nozzle.

Perhaps the inner part of the tower floor could be printed in the secondary material - the brim and first layer under the tower ExtA, but the circle WITHIN the tower could be ExtB?

Ghostkeeper, 2019-11-07:
The inner part of the tower floor was chosen on purpose to be the same as the rest of the first layer to contribute more to adhesion of the prime tower. Otherwise it might fall over more easily.

Ghostkeeper, 2020-03-04:
We'll close this as it's expected behaviour.

closed 2020-03-04 by Ghostkeeper