Ultimaker/Cura #6535: Cura 4.3.0 and Creality CR-10S Pro with Tiny Machines firmware not working.

DrStrixx, 2019-10-14:
I bought the printer about 3 weeks ago and it connected with some difficulties to Cura, I had to restart Cura or activate printer in Cura to get it going.

I did update firmware from TIny Machines because I had some problems with leveling and that solved that, but now it is losing connection between printer and Cura. Printer worked fine for some days and then it started to lose connection. And now it is lost completely.
[cura.log](https://github.com/Ultimaker/Cura/files/3726522/cura.log)
    

**Application version**

**Platform**

**Printer**
Creality CR-10S Pro
Latest Firmware from Tiny Machines. 

**Reproduction steps**
Booting printer and Windows finds it then connect USB and printer plays a pling plong sound as it find it. 
Start Cura and it can´t open port (Com4) Baud 115200 or communicate with printer. 
Computer find the printer without any problems.
I need to load Marlin (Aurdino) to get it going every time and then it works.
As soon as I shut printer down its not working when I boot the printer up again.

**Actual results**
After Marlin is loaded it works fine!

**Expected results**
I hoped that loading Marlin fixed the bugg.

**Additional information**
I did change the baud rate in Marlin to 115200 from 250000.
I did read that that would be the way to fix it, but it is not working.
I hope you got all information you need.

Regards

Ghostkeeper, 2019-10-15:
Did you install the Arduino drivers when the Cura installer asked for it?

This problem is not occurring with the Ultimaker 2 or Ultimaker Original, last time I tried (which was last week somewhere). Something must be different in that firmware you're using. There is no way I can debug that from here without having your printer next to me, so in order to fix this I'd need to know what the printer needs Cura to do differently in order to connect to it...

In the log file it shows that it's trying to connect to COM4 indefinitely.

DrStrixx, 2019-10-15:
Crap.. To big file. Hope this will work..

Hello and thanx for speedy reply!

What the printer needs Cura to do I do not know...More then connect solidly
every time... Without having to load Marlin or Tiny Machines firmware every
time I need to use the printer.
The question is why is it not communicate over Com 4 with baud set on
115200 or 250000. There should be no problem with that I think, but
something is wrong.

Did uninstall Cura and reinstall it with Arduino activated. Was not sure if
I did it...

And still it makes no connection with printer after reboot of printer!
If I let USB stay connected after I shut down the printer, LCD loading
screen is still on! And after I switch on printer (USB connected) LCD hangs
in loading screen!
But for some strange reason I can upload firmware to printer true Cura! And
send a print to printer, but it did malfunction, slamming the printer head
in to print-board and made a big hole in the magnetic print-board!! Had to
kill power to printer as fast I could. And LCD is stuck in loading screen.
After reboot of the printer the LCD is still stuck in loading screen! I was
forced to reload Tiny Machines firmware to make it work again! Not even
Marlin made it work. I did disconnect USB before I try to boot it up!
I boot up printer and connect USB. No connection. Upload Marlin and it
works..... For how long... Until next reboot of printer?

I made a .png of some settings in windows and Cura. Hope it helps you.
As you see I did attached some files to that I found...

Hope you have what you need to help me.
I will try out Creality latest firmware and LCD firmware to see if it
connects better to printer.

Regards
Christer

DrStrixx, 2019-10-15:
Hello again,

I did noticed that if I power of the printer with USB plugged in and not
power off the computer and then power on the printer again and waited about
90 sec the printer booted up and after another 65 sec from starting Cura
the Cura finds it! I never did expect that long boot time. I was thinking
something is wrong with Cura and printer communication.
So I patience(?) is what I need have, not stress away and think I must
hurry up 🚑... Then I can use it 👍 as it is. But if you can find a answer
to why it takes so long time, next firmware can solve that?

I hope this will work for me for now. And if I run in to more problems with
Cura and my printer firmware, I hope I could contact you contact you again

Many regards
Christer

Den tis 15 okt. 2019 kl 19:20 skrev Christer Arkemyr <drstrix@gmail.com>:

DrStrixx, 2019-10-16:
I did get an answer from Creality!
They recommend version 15.04.3! 4.3 is not working according to them. I
shall test that version. Old crap, but...

Regards
Christer

Den tis 15 okt. 2019 kl 23:26 skrev Christer Arkemyr <drstrix@gmail.com>:

Ghostkeeper, 2019-10-17:
> I made a .png of some settings in windows and Cura. Hope it helps you.
As you see I did attached some files to that I found...

Attachments through e-mail replies don't get posted publicly. You'd need to go to https://github.com/Ultimaker/Cura/issues/6535 to post an attachment.

DrStrixx, 2019-10-17:
[Attatchment.zip](https://github.com/Ultimaker/Cura/files/3739260/Attatchment.zip)
Okidoki

Ghostkeeper, 2019-10-17:
Cura is detecting a baud rate of 115200 rather than 9600 what you have in Windows. What happens when you change the configuration in Windows?

DrStrixx, 2019-10-17:
It change it back to 9600! I have changed to 115200 and it will not keep it..I will upload a pic

[image]

DrStrixx, 2019-10-17:
Did change it in another place (Right click on - PC - Manage)
[image]

Ghostkeeper, 2019-10-18:
Hmm, maybe @Microsoft could take a look at this? :smile: 

Do you know if other printer connectivity is able to connect? Like Pronterface?

DrStrixx, 2019-10-18:
Strange that old CURA 1.15 works, but not the newer one. 
I did test the Pronterface. Looks like it connects.
[image]

DrStrixx, 2019-10-23:
Any news?
I still use CURA 4.3 only thing is that I have to load Marlin and then Firmware every time I use it.

Ghostkeeper, 2019-10-24:
There is no real way in which we can debug this for you. We're sort of guessing in the dark, since it works fine with the Ultimaker 2.

DrStrixx, 2019-10-25:
Ok. That´s sad, but I can still use CURA and that´s is nice. A good software for an newbie in 3D printing!

closed 2019-10-25 by DrStrixx