Ultimaker/Cura #6536: Dual monitors can drop down menus

davisdesigns, 2019-10-15:
If I run Cura on my second monitor the drop-down menus don't work correctly the first part disappears in the second part stays open very difficult to select anything dragging the program back into the first monitor immediately resolves the issue.

rburema, 2019-10-15:
Hi @davisdesigns 

- Which version of Cura are you using?
- Which OS/flavour are you using? (If Linux, are you using the AppImage?)
- Are your video drivers up to date?
- Which video-card is used to run Cura? (Some people have a setup with onboard Intel + Nvidea.)
- Are your monitors connected via a docking station?

davisdesigns, 2019-10-15:
4.1
Win7 64
Yes 
Nvidia Solidworks compliant  card
Connected at card

Another issue is i have no "solid" option in the drop down only wire and x-ray?

rburema, 2019-10-16:
> Another issue is i have no "solid" option in the drop down only wire and x-ray?

'Solid' is available in the 'Prepare' stage, not the 'Preview' one.

--

Ok, there seems to be nothing that points to an existing issue given those answers.

- Does this happen for specific drop down menu's (like, only the settings and colorscheme-selection), or is it all of them?
- Can you attach the logs to this issue? They're in `%appdata%/cura/4.1/`.

davisdesigns, 2019-10-16:
I wish Solid was available during prepare/ slicing as wire view makes details difficult to see in cross section.

I'm not at work so i can't check but some menus were worse than others, I think the more content the flakier it would act, like playing whack-a-mole trying to select something.. lol 

I'll see if I can find the file you are referencing. 

I haven't had time to play around with it but I'm kind of wondering if having SolidWorks running in monitor 1 at the same time is what might be causing it. I'll check that too, honestly I wasn't even expecting a response. 

Thank you by the way for offering Cura for free I appreciate it.  I think the biggest pet peeve I have with it is pan/move are not the same as SolidWorks it makes it really difficult going back and forth manipulate the model.  It would be cool if that was an option to change how you manipulated in Cura to mimic other CAD programs.

davisdesigns, 2019-10-16:
It's the dropdowns with arrows that have the issue  in monitor 2 only.
Shutting down SW didn't help.

%appdata%/(roaming)/ cura/4.1/   

[stderr.log](https://github.com/Ultimaker/Cura/files/3736738/stderr.log)

davisdesigns, 2019-10-16:
[image]

davisdesigns, 2019-10-16:
Could this be from a duel desktop setting ?   If so let me know...   I use Nvidia control panel/ desktop manager.

Ghostkeeper, 2019-10-17:
Those dialogues are created by Windows, not by Cura. We might be looking at a driver issue here.

closed 2020-06-19 by no-response[bot]