Ultimaker/Cura #6538: Infill layer Height

MichelCHAVOT, 2019-10-15:
[image]
I use an infill layer thickness equal to 2 times the layer thickness.
This works very well, except if the infill density equals 100%. Is that normal?
Thanks

Michel CHAVOT.

rburema, 2019-10-16:
Hi @MichelCHAVOT,

100% Infill _always_ defaults to the line pattern, as it's supposed to be solid. This is expected behaviour.

cheers, Remco.

closed 2019-10-16 by rburema

ninovanhooff, 2019-10-16:
The Cura Marketplace provides a SettingsGuide plugin with visual explanation. 
It mentions that some layers can be combined, which may explain why there is infill in layer 10 for 100% but not for 99%.
See
https://github.com/Ghostkeeper/SettingsGuide/blob/d954be25b30c1877ec7ce75a0b233367a2a46a13/resources/articles/infill_sparse_thickness.md

Ghostkeeper, 2019-10-17:
Instead of infill it'll print everything with skin then. That tends to produce a stronger part since there will be no border any more between infill and skin.