Ultimaker/Cura #6540: Schema for fdm_materials XML

Sassw, 2019-10-16:
Recently when using the REST API to add custom materials to my S5 I was frustrated at the lack of documentation of the XML used for fdm_materials.  

https://github.com/Ultimaker/fdm_materials/blob/

It's now especially important to have this information since the S5 with Material Station will refuse to print some profiles unless certain information is contained within them.

I'm sure this information has been given to material suppliers to test and update profiles but it should also be available for end users to do themselves.

Can the schema for the XML be documented somewhere and also describe which fields are locking material use out of the Material Station.

ChrisTerBeke, 2019-10-16:
You mean https://github.com/Ultimaker/fdm_materials/blob/master/scripts/fdmmaterial.xsd?

Sassw, 2019-10-16:
woohoo!

Sassw, 2019-10-16:
woohoo!

closed 2019-10-16 by Sassw