Ultimaker/Cura #6543: Slicing calculation starts always

tmirante, 2019-10-16:
**Is your feature request related to a problem? Please describe.**
slicing stop

**Describe the solution you'd like**

the possibility to start and stop slicing calculation
**Describe alternatives you've considered**

wait for calculation a long time before the answer
**Affected users and/or printers**

every people using models with many details requiring a long load time and a long slicing time expecially during positioning  
**Additional context**

ChrisTerBeke, 2019-10-16:
You can disable auto-slicing in the preferences window.

fieldOfView, 2019-10-16:
There's also a neat plugin in the Marketplace that adds a pause/resume button to the slice area.

closed 2019-10-17 by nallath