Ultimaker/Cura #6545: Support Infill Line Directions with thickness greather than layer height

FerT74, 2019-10-16:
**Application version**
4.3.0

**Platform**
Windows 10

**Printer**
Makerparts XS

**Reproduction steps**
When you set these parameters in any model: 

Layer Height 0.1
Generate Support ON
Support Infill Layer Thickness 0.2 (doubles layer height)
Support Infill Line Directions [0,90]

Supports are printed in only one direction (0º in this case). The workaround is to set Directions as [0,0,90,90]. It seems that directions are evaluated per wall layer, not per support layer.

**Actual results**
Supports in only one direction

**Expected results**
Crossed lines supports

Ghostkeeper, 2019-10-22:
So the reason why this is happening is because the Support Layer Thickness setting causes Cura to "skip" a layer of support and then extrude twice as much in the next layer instead. This causes one of the directions to also skip. The same problem also exists for infill, not just support.

However when considering a fix for this we ran into an issue: The skipped line directions ARE in fact used, because the layers are not skipped if the double layer height doesn't fit. It'll use the single layer height then. That means that at the top of the support area there is still a layer that must get some line direction. Especially there it's important that the line direction is perpendicular to the previous one so that the overhang is correct.

We can consider doubling the Directions automatically to make the behaviour more as the user expects it, or leaving it as is to make the behaviour more powerful. What do you think?

FerT74, 2019-11-08:
Is it possible to change direction based on a "support layer counter" instead of the "main layer counter" (I think there is a global layer counter now). So, the change in direction would be per support layer, regardless the layers skipped.

Ghostkeeper, 2019-12-03:
What I was saying is that the layers are not always skipped. Here's an ASCII illustration of how your support could look with a certain shape. `|` is model layer, `-` or `=` is support for single layer thickness resp. double layer. `_` is the build plate. Layer numbers on the right.

```
|||||||||||||||| (5)
||||||||||||     (4)
||||||||----==== (3)
||||    ----|||| (2)
----====|||||||| (1)
________________
```

It will only place down double-thickness layers where the support fits on both layers. If it doesn't fit, then a single-thickness layer is placed down. In this case that happens at:
* Layer 1 on the left side, where the space between the build plate and the model is only one layer.
* Layer 2 on the right side, because it's finishing the thickness of the double-thickness layer that goes over the rest of the layer, since here there is model beneath it.
* Layer 3 on the left side, where there is model directly above it.

So the question is, what happens to the line directions of layer 2 in this case?