Ultimaker/Cura #6547: Generating support causes random shaped support "bone spurs" to print adjacent to model.

mbmast, 2019-10-16:
Even with very simple models, Cura 4.3.0 adds strange looking, spiky support adjacent to the model.  The photos included here have brim generated, but the problem occurs even with no brim.
[image]
[image]
[image]
[image]
[image]

Here's the model file:

[Support Test.zip](https://github.com/Ultimaker/Cura/files/3736590/Support.Test.zip)

nallath, 2019-10-17:
I think this might be caused by a rounding error. At the very least I can already provide a workaround for this. If you set the `support horizontal expansion` to -0.1, they weird areas will no longer be generated.

Ghostkeeper, 2019-10-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/4219

closed 2019-10-17 by Ghostkeeper

mbmast, 2019-10-17:
@nallath's suggestion to set `support horizontal expansion` to -0.1 works.  If this suggestion does NOT provide a workaround to issue #4219 (and I suspect it won't) then I would say this issue is NOT a duplicate of #4219.

Ghostkeeper, 2019-10-18:
Setting horizontal expansion to -0.1 also improves it for the model in #4219. It's not completely gone then though. Seems that it needs -0.2 there. But the issue is very similar.