Ultimaker/Cura #6548: error support_z_distance = 0.0. (cura 4.3)

Goodfeat, 2019-10-17:
Hello, I found a bug in CURA 4.3
CURA does not generate support if the parameter support_z_distance = 0.0.

[image]

if the support_z_distance > 0, then everything becomes normal

[image]

-Windows 10
-Cura 4.3
-Delta printer

rburema, 2019-10-18:
Hi @Goodfeat,

If I just follow the reproduction steps it doesn't reproduce for me. Could you attach the project (saved, not exported, you may have to zip it to get github to accept it), please?

cheers,
Remco.

Goodfeat, 2019-10-18:
[альтернатива 0.zip](https://github.com/Ultimaker/Cura/files/3742822/0.zip)
This is an archive with the stl file. I use two nozzles. the second nozzle prints only the bonding layer

rburema, 2019-10-18:
I see.

A workaround is to put either the stair step width or height to a low value.

Maybe related to #6185 which was originally closed as a duplicate, but the fix for the underlying issue was reverted at one point (see #6163 ). Stair-stepping is a piece of code we've had/realised some problems with over the past year or so.

Ghostkeeper, 2019-10-18:
Could you save a project file instead of exporting the STL file? To save a project file you'd need to go to File -> Save. The resulting file not only contains the 3D model but also all of the settings that were used. This way we can reproduce the problem and see what is going wrong with it.

I've also tried to do this with a basic overhanging model but it just generates support as well as support interface fine as expected.

rburema, 2019-10-18:
(Also, while there's probably a reason you do it this way, I thought I'd mention that it may be better to print that model upside down.)

Goodfeat, 2019-10-18:
> (Also, while there's probably a reason you do it this way, I thought I'd mention that it may be better to print that model upside down. so no support is needed.)

Not in this case. I just provided a model by which to reproduce the error.
I am testing printing with two nozzles with plastic PLA

rburema, 2019-10-25:
Seems to happen only on deltabot type printers. If I have a CustomFFF printer with elliptical buildplate and origin at center it doesn't happen. I'm not sure _why_ because there don't seem to be much other parameters different between those situations that could influence this. Perhaps I've overlooked something though.

rburema, 2019-10-25:
Please open a new issue, so we can, for example, open & close them separately.

Goodfeat, 2019-10-25:
Ok

rburema, 2019-11-12:
I've some time to further investigate now, sorry for the wait.

It seems I _can_ reproduce it (on a non-deltabot printer) if I set the layer-height to 0.2 (like in the original). This also happens to the stair-stepping height. Putting the 'support stair step height' to anything below 0.2 will work as a workaround.

rburema, 2019-11-12:
I now think this is a duplicate of #6379 for which we have an internal ticket already (CURA-6808).

Goodfeat, 2019-11-15:
I looked at the problem # 6379,
There you mentioned that the problem is "Support stair step height". 
This topic is about a problem with the parameter "Support Z Distance"
interesting that the problem is very similar

rburema, 2019-11-15:
Yes, but your problem will also go away if you put 'support z distance' to a smaller problem, regardless of what 'support top distance' is. These values interact with each other.

Goodfeat, 2019-11-15:
Really. the problem disappears
This is a "bug" or "particularity" when building supports ?

rburema, 2019-11-15:
This is a bug within the stair-stepping. This is a bit of code we've had trouble with in the past. It seems each time we fix a bug there, another one appears. Fortunately this last one has an easy work-around.

closed 2019-12-03 by Ghostkeeper

rburema, 2020-06-25:
When the time comes (it might be a while) could you check again in 4.7? It seems we finally might've fixed this issue in the master branch.