Ultimaker/Cura #6556: [4.2.1] hangs on save to file

TestniTester, 2019-10-19:
**Application version**
4.2.1

**Platform**
Ubuntu 18.04.2 LTS

**Printer**
Ender 3

**Reproduction steps**
load any stl, click slice, click save to file

**Actual results**
 it hangs instantly on click of button

**Expected results**
popup for file location to save should be seen

**Additional information**
everitihng was working normaly didnt change anything (that i know) and it just stopped!?
Downloaded the last version 4.3.0. same error

**LOG**
2019-10-19 12:53:49,754 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [646]: Slicing took 44.96093535423279 seconds
2019-10-19 12:53:49,755 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [647]: Number of models per buildplate: {0: 1}
2019-10-19 12:53:49,755 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [661]: See if there is more to slice...
2019-10-19 12:53:49,760 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [179]: Socket state changed to Closing
2019-10-19 12:53:49,764 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [201]: Socket debug: Arcus Error (13, native 11): Closing socket because other side requested close.
2019-10-19 12:53:49,768 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [181]: Socket state changed to Closed
2019-10-19 12:53:49,770 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [806]: Backend quit with return code 0. Resetting process and socket.

Ghostkeeper, 2019-10-21:
We've seen this problem on Windows before. See this bug report: https://github.com/Ultimaker/Cura/issues/1537 . There the issue is that the file dialogue that Qt creates is created from all of Windows' elements including its Explorer extensions (like e.g. Dropbox that adds an entry on the left side).

It's never been reported on Linux so far though. So perhaps this issue is different. But perhaps it could be related to some other software you've installed that just causes the file dialogues on your system to freeze.

I'm not able to reproduce the issue on Ubuntu 19.04, so far.

ITeach2Wheels, 2019-10-23:
I"ve been slicing files all day today with no issues in 4.2(?).  I updated to 4.3 tonight and am having a similar issue.  The file downloads, slices and appears to save normally (ish...it asks for a save location, I choose it and the save screen disappears)  I seem to recall more than a disappearing screen before 4.3.  There is no record of the file being saved.  My workflow is exactly the same.

I have reinstalled once with no change.

ITeach2Wheels, 2019-10-23:
Sorry for the extra post.  The issue seems to be with the Anycubic Linear Plus profile.  I've tried custom and standard filament profiles with the same result.  When slicing for my i3Mega the file saves normally.

TestniTester, 2019-10-23:
OK, i dont know how is this possible, but now (after coupple of days of not trying) its working OK!?
So i think it was something on my OS, dont know what, becouse i  have restarted the computer several times when testing.

TestniTester, 2019-10-23:
Im gonna close this, becouse i dont think its a bug

closed 2019-10-23 by TestniTester