Ultimaker/Cura #6559: [4.3] Cura prefixing temperatures before start code

Wireheadbe, 2019-10-20:
When slicing with cura, it seems the temperatures are added before the "Starting G-Code" (using ender 3 profiles).

e.g.

1. ;FLAVOR:Marlin
2. ;TIME:5982
3. ;Filament used: 4.21699m
4. ;Layer height: 0.2
5. ;MINX:72.327
6. ;MINY:90.112
7. ;MINZ:0.2
8. ;MAXX:139.799
9. ;MAXY:129.886
10. ;MAXZ:48
11. ;Generated with Cura_SteamEngine 4.3.0
12. M140 S45
13. M105
14. M190 S45
15. M104 S200
16. M105
17. M109 S200
18. M82 ;absolute extrusion mode
19. START_PRINT

I'm using Klipper, and have my own macro's defined (START_PRINT) - e.g. I put the probe in the middle of the bed to heat it up with the bed, as to have no thermal drift when mesh probing. However, Cura prefixes it's temperature settings for the print _before_ the start code. - So my own macro only kicks in after Cura set the temperatures.

I would expect Cura to do that behind the start code. Or is there some option I'm overlooking?

Kind regards,
Jeffrey

fieldOfView, 2019-10-20:
Cura checks if there is any preheating going on in the start gcode. If there's no preheating done, Cura instructs the CuraEngine to insert a preheating sequence to prevent cold extrusion.

In order to thwart this safety feature, make sure your start gcode snippet contains the strings `{material_print_temperature_layer_0}` and `{material_bed_temperature_layer_0}` somewhere. It could be in a comment, or in an actual preheating sequence.

Wireheadbe, 2019-10-20:
Ah awesome, many thanks. I couldn't find this directly in the docs. Many thanks!

closed 2019-10-21 by nallath