Ultimaker/Cura #6560: Deliberate over-width lines

Asterchades, 2019-10-20:
**Is your feature request related to a problem? Please describe.**
One of the few elements of Cura that I've always had a small issue with is the distinct separation of visible lines. Most of the time this represents merely an aesthetic issue where gaps may be visible between skin lines or between walls, but sometimes it can result in the walls "delaminating" due to insufficient horizontal adhesion between lines after the extrusions shrink slightly during cooling.

Note that I don't consider this a bug. If anything, it's the result of excessive accuracy - lines are printed at the specified width and spaced exactly according to said width. This leaves zero deliberate horizontal merging of the lines, relying instead on vertical merging to hold everything in place.

**Describe the solution you'd like**
Following recent experiments, I've all but removed this issue by calculating (fortunately by way of a custom profile) the line extrusion multiplier such that the lines for the skin (top/bottom), roof (top surface), wall_x (inner wall/s), and layer_0 (initial layer) are wider by roughly 10.73% of the layer height. The formula I've used is only a rough approximation, however, based on the following:

material_flow * (1 + (layer_height / (<feature line width> + layer_height) * 0.1073))

Due to the pill-shaped nature of lines, this will actually result in slightly less over-extrusion than I'd like, but it's pretty close.

Since Cura must already include the appropriate formula to calculate the required flow for a given line width based on the nozzle size, it seems like it would be easier (at least from an end-user perspective) to simply allow the lines in question to be extruded at the slightly higher width without any additional spacing (such as you'd get by simply increasing the line width with the current options). Because this is a pre-calculable amount of additional width which doesn't change regardless of the line width or where it's used, it could simply be a checkbox-type option rather than a manual distance entry.

As an added bonus, this makes the skin overlap parameter redundant, since both the skin and inner wall/s will be better merged into each other.

NB This extra flow/width is not applied to the outer wall or infill. The outer wall is left untouched to maintain dimensional accuracy, while the infill is left untouched as it is superfluous. This does mean the infill overlap percentage is still required.

**Describe alternatives you've considered**
The immediate alternative is, naturally, simply adjusting the extrusion multiplier as I've done. I feel this is less accurate with my formula, and a duplication of efforts in the event a more accurate formula is used.

Another alternative would be deliberately reducing the spacing between the lines by a corresponding amount, shifting the inner wall/s closer to the outer and the skin lines closer together (as well as extending them proportionally to the inner wall). This would increase print times very slightly, however, though it would result in a very slightly more uniform surface due to the outside wall being of equal thickness to all other visible lines, as well as avoiding the issue with the outer wall not quite merging as much without other settings being altered.

**Affected users and/or printers**
This should affect any and all FDM/FFF printers on any model being printed. The most obvious impact would be on flat surfaces but all details would benefit.

**Additional context**
The exact over-extrusion is based on the theoretical amount of overlap required to merge 50% of the mass of the rounded edge of a given extrusion into the next extrusion over, theoretically creating a 100% fill between the two extrusions. This is calculated based on the following, which I would ask that you check to make sure I haven't made a mistake:

[image]

This image represents the upper half of one side of the rounded edge of the extrusion - an area of 1/2 the layer height square - assuming it forms an ideal pill shape. The exact same will be mirrored vertically underneath it, while the rectangular shape of the extrusion will be off to the left, then another rounded portion to the left again of that.

The idea is that when the red portion of one extrusion overlaps with the red portion of another extrusion it should be pushed into the blue area. So if you can match the area of the red portion with the area of the blue portion, you should create a 100% fill - in theory. The end result is both a smoother top/bottom surface and better horizontal line adhesion, since now they're being merged into each other similar to how layers are merged. This also slightly increases inter-layer adhesion.

The area of the red portion is calculated by working out the segment area then subtracting the triangle as outlined by the green line. The segment area is "­­πr² * θ / 360". The area of the triangle is "r² cos θ * sin θ / 2", thus the area of the red portion is ­­"πr² * θ / 360 - r² cos θ * sin θ / 2". θ is the angle from the base in degrees, while the exact value of r is irrelevant as it is a constant.

The area of the blue portion is calculated by working out the area of the rectangle it borders on, then subtracting the segment as outlined by the green line and the triangle outlined by the green line against the base. The area of the rectangle is "r² cos θ". The area of this segment is "πr² * (90 - θ) / 360". The area of the triangle is the same "r² cos θ * sin θ / 2" as before. This makes the area of the blue portion "r² cos θ - πr² * (90 - θ) / 360 - r² cos θ * sin θ / 2". Again, θ is the angle from the base in degrees and r is a constant of irrelevant value.

Solving for θ when "πr² * θ / 360 - r² cos θ * sin θ / 2 = r² cos θ - πr² * (90 - θ) / 360 - r² cos θ * sin θ / 2" yields an angle of 38.242° for when the two coloured portions have equal areas (I used a pair of graphs for this and looked for where they crossed - I no longer recall how to solve trig functions properly). Using "cos 38.242°" yields 78.54%, meaning 21.46% of r must overlap to create the 100% fill. Since r is half the layer height, this means the overlap is 10.73% of the layer height.

Ergo, extruding the affected lines (top/bottom, top surface, and inner wall) at 10.73% of the layer height wider than their specified width or the same 10.73% of the layer height closer together (nudging towards the outer wall) should result in the void between the lines being filled completely.

Ghostkeeper, 2019-10-21:
There is a flaw in your logic: Cura never uses the pill shape model to compute the flow rate, but it doesn't use the pill shape to compute the distance between lines either. Cura assumes a perfect rectangle. It extrudes precisely enough material to fill the rectangle (in your graph indicated by the blue shape with the circle arc underneath it). And then the next rectangle gets placed immediately adjacent. This should already push the material in the red area upwards into the blue area.

Your proposed factor of 0.1073 times the layer height doesn't only depend on the layer height. It also depends on other factors, such as the ratio of line width to nozzle size (overpressure) and more importantly the temperature and viscosity of the material. As such, applying a static increase in material extrusion is not possible. Or rather, it's theoretically possible to model all material properties out as well, but there are a lot of factors involved including intrinsic properties of the material and environmental factors. That's why we didn't make the Flow settings a checkbox but rather a rational percentage.

Profile designers can enter a static amount in there that they calculate from the layer height (like you did) or measure from reality. They still can't factor in any environmental factors then but will at least have a better time to factor in temperatures, viscosity and pressure. *This way the user won't have to duplicate that work.*

Profiles can be applied to every level of configuration in Cura. We can change the global default, the default for a certain printer, the default for a certain printer+nozzle, the default for a certain printer+nozzle+material or the default for a certain printer+nozzle+material+quality. In this case, you'll probably want to configure this at the material level.

> Solving for θ when "πr² * θ / 360 - r² cos θ * sin θ / 2 = r² cos θ - πr² * (90 - θ) / 360 - r² cos θ * sin θ / 2" yields an angle of 38.242° for when the two coloured portions have equal areas (I [...]

Solving a formula like that can easily be done in Wolfram|Alpha! [Here is your formula.](https://www.wolframalpha.com/input/?i=pi*c%5E2+*+a+%2F+360+-+c%5E2+cos%28a%29+*+sin%28a%29+%2F+2+%3D+c%5E2+cos%28a%29+-+pi*c%5E2+*+%2890+-+a%29+%2F+360+-+c%5E2+cos%28a%29+*+sin%28a%29+%2F+2%2C+solve+for+a) This has multiple crossings, but one of the options is 38.24 degrees so that seems to be correct.

Asterchades, 2019-10-21:
There's always something I hadn't considered... but I guess that's how we learn, right?

So, realistically, I'm already duplicating effort as a result of trying to calculate what would be the rectangular volume of an extrusion from a pill-shaped base, when Cura is actually "skipping the middle man" and just going straight for the rectangle. As a result, instead of attempting to compensate for a perceived under-extrusion between lines I'm instead simply over-extruding them instead which produces the results I was hoping for only due to a staggering coincidence.

In this case, it's simply correcting a different issue which could be the result of any number of other factors - temperature, impurities in the filament, the actual composition of the filament, a slightly mis-sized nozzle changing the pressure, or simply uneven or restricted flow due to mechanical interference.

This does leave me wondering, though... would that behaviour influence attempts to calibrate the extrusion? Assuming you were to extrude a single wall and everything went theoretically correct, the resulting extrusion won't have flat side walls. A "perfect" 0.4mm line from a perfect 0.4mm nozzle using rectangular flow would likely come out slightly wider than 0.4mm after it cools, even considering PLA shrinkage. In which case, am I simply trying to compensate for an issue I created myself by adjusting the flow percentage to make such a line be equal to 0.4mm?

After all, I "tuned" my extrusion multiplier to 93.5% based on a two-wall object and some calipers, making sure that two 0.4mm walls made 0.8mm. This is in addition to ensuring the extruder itself has the correct step/mm settings. But if these walls were deliberately a fraction over-sized, all I've done is introduce under-extrusion. This seems even more likely considering the "corrected" extrusion multipliers my equation gives are around 100%.

Which leads me to one last question: Is there an appropriate way to objectively calculate or measure the extrusion multiplier, or is that something which is based on subjective viewing only?

Incidentally, I used [Desmos' Graphing Calculator](https://www.desmos.com/calculator) to find my cross-over point. I'd heard of Wolfram but always though it was a paid/subscription service.

Ghostkeeper, 2019-10-22:
> This does leave me wondering, though... would that behaviour influence attempts to calibrate the extrusion? Assuming you were to extrude a single wall and everything went theoretically correct, the resulting extrusion won't have flat side walls.

Yes, you're right there. The extrusion that CuraEngine produces is exactly correct for a rectangular line, which should be correct for an infinite amount of adjacent lines (practically: for skin). If you're printing a single line it'll be much wider.
One behaviour that we've also observed here at Ultimaker is that usually the first layer consists of alternating wide and narrow lines. This is caused by the first line being much too wide and then blocking the extrusion of the second line, causing underextrusion there, and allowing the third line to be wide again, and so on. So yeah, practice is always different I guess.

> Which leads me to one last question: Is there an appropriate way to objectively calculate or measure the extrusion multiplier, or is that something which is based on subjective viewing only?

I think weight of extruded filament is the most reliable objective measurement:
* Set Flow to 100% initially in Cura.
* Print an object that is representative for what you normally print.
* Measure its weight and compare this to the filament usage estimate that CuraEngine gives (listed in the top of the g-code). This filament usage is measured in length, so you'll need to multiply that by the cross-sectional area of the filament and its density. If you don't trust the diameter or density of the filament, you could take a sample by cutting off an equal amount of filament by hand using a piece of rope to measure length and weighing that.

This way you're directly measuring the amount of extrusion that the printer is commanded to give vs. the amount of extrusion you're actually getting. You can then compensate the flow to give the correct amount of extrusion. It's important that you use an object that is representative for the type of prints you'd like to optimise extrusion for. The extrusion is going to be different for every shape as well due to counter-pressure.

printingotb, 2020-03-05:
@Ghostkeeper Based from what I see in almost all facebook groups about 3D-printing, on some Youtube channels and blogposts, the vast majority of people believes that measuring a single line and "calibrating" flow based on it is the way to go.
Would it be possible to get an "official" guide or article about how you are supposed to calibrate flow in cura? (or how we shouldn't do).
Either directly in Cura or a guide that I can link to when I support people?
This seems to be one of the largest myths in 3D-printing using Cura as the slicer.

Ghostkeeper, 2020-03-09:
I disagree with that method, especially because a single line is always much wider than multiple adjacent lines. If you adjust the flow based on that, you're going to get some underextrusion.

A bit of underextrusion is actually beneficial for visual details though, so optimising for single layer line width is not always such a bad thing.

I've found it more useful to optimise for weight instead. But I'm just one person and it's not even my job to print 24/7 or to optimise profiles. There's people out there with much more experience than I. I print maybe 3 or 4 things a week (for the past 5 years).

Writing an official guide falls within the scope of the Settings Guide (could be under "Line Width" or "Flow"). But that same guide is advising you to tackle the original problem that causes the overextrusion/underextrusion rather than adjust the flow to compensate for it.

printingotb, 2020-03-09:
I simply calibrate E-steps as precise as I can and set flow to 99%, or a couple % lower. I'll try weighing as well, might give me the final flow accuracy. I fine tune a lot of profiles for my own production, I haven't needed to reduce flow more than that while still getting the best possible surfaces. 

Hmm, perhaps measuring the width of a single line and change flow only for outer wall lines is the best of two worlds.

Would something like this work to show what happens when flow is changed? (Drawn in Fusion360)
It could perhaps also mention the horizontal expansion setting as a way to compensate for the wider bulge on outer lines.
[image]

Ghostkeeper, 2020-03-13:
I think that image is a pretty good diagram for how the flow would behave in a perfect world. The bulge gets pushed out on one side if there is space.

There are still some differences with reality though, and they depend on which wall gets printed first. Obviously with the 100% flow, it won't actually flow into perfectly rectangular beads. It will still bulge. And then the next wall besides it gets pushed out wards. This is why "Outer Before Inner Walls" is a huge factor on dimensional accuracy when using 100% flow. And like I've mentioned before, one observance that we've made is that this also causes the second bead to have more back-pressure, which reduces the total amount of material for that line.

The bulging is also much more extreme than what you've drawn at 100%.

GregValiant, 2026-09-10:
@Asterchades can this be closed? (I'm bored and doing housework).

closed 2026-09-10 by GregValiant

Asterchades, 2026-09-11:
By all means. Ultimately it would seem that this was just a mathematically (over)complicated way of solving the "look and see" approach to proper calibration, which I hadn't really learned at the time I made the post.