Ultimaker/Cura #6562: Missing settings in 4.3.0, unable to set Z Seam Position with "User Specified"

pnt103, 2019-10-20:
**Application version**
Cura 4.3.0 Win64

**Platform**
Microsoft Windows 7 Pro 64-bit

**Printer**
Ender 3 Pro

**Reproduction steps**
With all settings visible (according to the Settings Visibility), loaded STL file, set all settings to defaults except Z Seam Location, which was set to User Specified.

**Actual results**
Z Seam Position appears greyed out, set to "Back", and cannot be changed.  "Z Seam X" and "Z Seam Y" do not appear at all, although they are ticked under "Settings Visibility".

**Expected results**
I should be able to at least change "Back" to "Front", or "Front Left", etc.

**Additional information**
I've just encountered the same problem as Roboman10 in Issue #6414 after upgrading one machine from 4.2.1 to 4.3.0.  This PC does not show "Z Seam X" and "Z seam Y" lines under "User Specified".  It does show "Back" but it's greyed out and when I hover over it the help text says "This setting is not used because all the settings that it influences are overridden ... Affects Z Seam X    ZSeam Y".  An identical PC also shows only "Back" but does allow me to set Front, Front Left, etc, though not X/Y coordinates (it doesn't show them either).  Both are used with Octoprint and an Ender 3 Pro, both have the same extensions/plugins, and both have all settings turned on under Settings Visibility.  Could this be to do with how I installed the upgrade, specifically whether or not I elected to keep previous config?

smartavionics, 2019-10-20:
Try closing the settings section and opening again. That works for me when the X/Y fields are not visible.

pnt103, 2019-10-20:
Alas, not me.

Ghostkeeper, 2019-10-21:
Let's continue here: https://github.com/Ultimaker/Cura/issues/6414

I think the author accidentally closed his ticket.

closed 2019-10-21 by Ghostkeeper