Ultimaker/Cura #6568: Where is "maximum z speed"?

marvinmarmelade, 2019-10-21:
Hello fellow humans,

I wanted to increase the "z hop speed" because it was moving endlessly slow but i can not increase the value over 3mm/s (field turns red and doesn't let me slice the damn thing). I found out that the "maximum z speed" option was to increase but I CANT FIND THAT OPTION. Where did it go or am I just blind or something? I am using the newest Cura 4.3.0.

Pls help me...

rburema, 2019-10-22:
Hi!

'Maximum Speed Z' is not normally visible, it's defined per machine as a maximum safe upper limit for that specific machine. You'll have to adjust the profile if you want that setting to change.

You can either directly edit the definition itself (in the install folder, not recommended), or add a definition (in the %appdata% folder, or equivalent for your OS) that extends the one you're working from and overwrite that specific value. 

cheers, Remco.

P.S. Do you have a 'tevo tarantula'? That't the only profile that has 3 mm/s by default as far as I can see.

Ghostkeeper, 2019-10-22:
Basically, Cura is saying here that your firmware limits the Z speed to 3mm/s. That's the information that [Cura has](https://github.com/Ultimaker/Cura/blob/985aac9e601337d310b5a9c3e9160ed39852104e/resources/definitions/tevo_tarantula.def.json#L44).

creavitt3d, 2019-10-24:
Without the "maximum z speed" setup, my Robo 3D R1+ stop to print. Because this issue i returned to Cura 4.0 and work again. Have anybody a recommendation to solve this issue?

Ghostkeeper, 2019-10-24:
What is your Z Hop Speed set to?

creavitt3d, 2019-10-24:
in Cura 4.0 the default value is zero, its caused problems. i change to 5 mm/s the the problems are solved. i dont have more the value in 4.3 version. i'd needed downgrade to 4.0 again!

Ghostkeeper, 2019-10-25:
In Cura 4.0 there was no Z Hop Speed yet. What is the setting called "Z Hop Speed" set to in version 4.3?

creavitt3d, 2019-10-25:
Sory, I don't remenber the value. I don't changed the "z hop" standart cura 4.3 installation value. I had  Installed cura 4.3 and used the standar robô 3d R1+ configuration.

I preserved the fail gcode archive, if you need i can send to you.

Ghostkeeper, 2019-10-30:
I don't really need the g-code. I can believe that it's happening for you. But to figure out why or how it happens, I'd need to see a project file. Z Hop Speed was just a hunch. For a project file, you'd need to be able to reproduce the problem.

With the Robo 3D R1 printer, Z hop is disabled by default. When slicing something, it seems to be making the Z movements at 10mm/s (`F600`).

creavitt3d, 2019-10-30:
OK, in attached are the "fail gcode" (Cura 4.3) and the "fully printed" gcode (Cura 4.0 after max z adjust)" to you analysis. For me its very important to understand the issue to continue upgrade the software.

[SUP.zip](https://github.com/Ultimaker/Cura/files/3790407/SUP.zip)

Ghostkeeper, 2019-10-31:
Like I said, **I don't really need the g-code. I can believe that it's happening for you.** We'd need a project file, which is not a g-code file. The project file can be saved by clicking on "File" in the top menu bar and then "Save...".

However I also don't see anything in your fail g-code that would make your printer stop. You may be running into a firmware bug here. But to report that you'd need to know what g-code command is causing it to stop.

creavitt3d, 2019-10-31:
Attached are the project files .3mf

[Sup.zip](https://github.com/Ultimaker/Cura/files/3796098/Sup.zip)

Ghostkeeper, 2019-11-01:
So your Z Hop Speed (under the "speed" category) was set to 10. I've set it to 5 for you:

[R3DR1_SupCURA4_3_5mms.zip](https://github.com/Ultimaker/Cura/files/3797223/R3DR1_SupCURA4_3_5mms.zip)

creavitt3d, 2019-11-01:
OK, i will use 5 mm/s in "Z Hop Speed" in Cura 4.3 config. Thanks

closed 2019-11-04 by Ghostkeeper