Ultimaker/Cura #6572: Crash on opening materials settings when UM2 in printer list

atrophy, 2019-10-23:
**Application version**
4.3.0

**Platform**
Windows 10 Enterprise version 1809

**Printer**
Ultimaker 2

**Reproduction steps**
 - Clean installation of Cura 4.3.0 after removing application data folder manually
 - Add UM2 printer at startup prompt
 - Open Preferences -> Configure Cura
 - Click on "Materials" in the side menu

**Actual results**
Application hangs for 5-30 seconds and closes without prompt.

**Expected results**
Open the materials preferences to add or edit materials.

**Additional information**
Only occurs when a UM2 printer is added, tried adding a UM3 in the first-run prompts and everything worked as intended. I then added a UM2 in the Preferences -> Printers area in addition to the UM3 and the crash reoccurred.

rburema, 2019-10-24:
Seems like it doesn't need to be a totally clean install (I didn't select OlsonBlock): I got the same behaviour when I tried it on an already open Cura 4.3 with a few printers added.

We've changed a lot of those systems though, so I've checked if this occurs on master (which is all going to be in 4.4 since we didn't have feature freeze + version-branch yet) and it adds it goes into materials normally after the addition.

(Also might be a duplicate, I seem to remember on of my colleagues fixed an issue with the UM2 recently.)

Ghostkeeper, 2019-10-24:
Yes, this works in Master. We've done a big refactor that made the MaterialManagementModel more robust with this sort of thing. Thanks for reporting though!