Ultimaker/Cura #6573: How to make .curapackage files?

maukcc, 2019-10-24:
Is there a way to manually make .curapackage files?

nallath, 2019-10-24:
The easiest way to do this is to upload them to contribute.ultimaker.com 

It will generate these packages even if you don't release them (so that you can test them locally if everything went right)

closed 2019-10-24 by nallath

maukcc, 2019-10-24:
This will not work as the package is not specifically for UM hardware :(
Please reopen this

nallath, 2019-10-24:
So when you upload the package it immediately says that it must have UM hardware in it? 

Once you upload a package, you can click the SDK version numbers to download the package.

maukcc, 2019-10-24:
Ahh, got it. click on the SDK to download.
Thanks !!

Ghostkeeper, 2019-10-24:
For posterity, the .curapackage file is just a .zip archive so you can edit the contents with an archiver application.

maukcc, 2019-10-24:
We can not edit or delete files that are in the curapackage file. We can add files and edit and delete the added files.