Ultimaker/Cura #6576: Horizontal Expansion - no minus sign

RegRayRat, 2019-10-24:
**Application version**
4.3.0

**Platform**
Windows 7

**Printer**
Tevo Tarantula

**Reproduction steps**
Trying to apply a minus 0.05% value in Horizontal Expansion box

**Actual results**
no minus sign so unsure if minus value applied

**Expected results**
Earlier versions showed -0.05 (minus sign)

**Additional information**

nallath, 2019-10-24:
I can't reproduce this. I've tried to do this on 4.3 and from the latest master and I can type in negative and positive numbers for horizontal expansion. The only difference that I can see is that I'm trying this on Fedora (linux)

RegRayRat, 2019-10-24:
Thanks for quick reply - I only have Windows OS machines. I'll install Cura on a laptop (Win 7) and try. I'll report back.

fieldOfView, 2019-10-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/6277 (which unfortunately does not have a workaround yet)

closed 2019-10-24 by nallath