Ultimaker/Cura #6578: [4.3] Can't select what printer to use for printing

Quasardian, 2019-10-25:
**Application version:**
4.3.0

**Platform:**
Windows 10

**Printers:**
Ultimaker S5 (3x)

**Reproduction steps:**
Have multiple printers connected to the cloud
Print over network
No selection screen to choose what printer to use

**Expected results:**
By default have the option to select what printer to use.
Possibility to disable it all together in the settings.

**Additional information:**
This function is vital for our way of using the printers, some others on the forums are complaining about this issue as well. For the time being we had to roll back to Cura 4.2.

ChrisTerBeke, 2019-10-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/6433.

closed 2019-10-25 by ChrisTerBeke

Quasardian, 2019-10-25:
@ChrisTerBeke Thank you I wasn't able to find a duplicate.